LaserWeb/deprecated-LaserWeb1 #2: Formatting on mobile

quillford, 2015-11-10:
The page works wonderfully on mobile. The gcode preview works with dragging and pinch to zoom. However, the formatting isn't quite there. 
[image]

openhardwarecoza, 2015-11-10:
Indeed I am aware of that.  I am not using a responsive layout (i was lazy and used two absolute <divs> left and tight..   I also still need to squeeze in the Webcam widget and CAM components, so screen real estate might be getting tight.   On my build of reprapweb I hide the less often used things in Modals that pop out from buttons. Will probably do the same here

openhardwarecoza, 2015-11-10:
[image]

@quillford, give this one a go on your tablet?  Moved the jog to a widget - plenty of space for macro buttons and the like where your motor button is

https://github.com/openhardwarecoza/LaserWeb/commit/623ad62ac55ce47e485edf933c04dcde1ccdbbf1

openhardwarecoza, 2015-11-10:
Fixed a bug.  Machine Control button should be disabled until a port is selected

https://github.com/openhardwarecoza/LaserWeb/commit/2d1ad4a2570646e2b667c5eefef4486d27f585e5

openhardwarecoza, 2015-11-18:
Oops accidentally clicked close.

This morning I upgraded to bootstrap 3.3.5 (from 3.1.1) and changed to a responsive layout with standard bootstrap formatting (no absolute divs)

Works on my Galaxy S5 and PC/ Please test on tablet etc and revert with comments

closed 2015-11-21 by openhardwarecoza