LaserWeb/deprecated-LaserWeb1 #4: Merge in Millcrum so we can get svg ->cut and dxf-> cut

openhardwarecoza, 2015-11-10:
Merge in Millcrum so we can get svg ->cut and dxf-> cut

quillford, 2015-11-15:
Is anyone working on this? If not, I'll start working on it in my fork.

openhardwarecoza, 2015-11-16:
I actually worked on it a bit:

openhardwarecoza, 2015-11-16:
Right so. I have made a bit of progress, but got stuck.  @andrewhodel my reverse engineering js skills (those ones that make you cringe, I know) has managed to get to the point where I open a file, it converts it to millcrum.  But my generate doesnt do what your generate does...

andrewhodel, 2015-11-16:
Basically it runs the millcrum code in an eval and uses the try catch block to catch any errors from the eval.

If you have the millcrum code, just copy and paste it directly in the generate block and don't run the eval statement.  Then you will be able to see line numbers for the errors in the JS console and work through any problems.

All of the millcrum code runs functions in inc/mc.js

openhardwarecoza, 2015-11-16:
aha! the Eval is the key! Didnt know what it does.

Okay let the digging proceed.

openhardwarecoza, 2015-11-16:
i see so var tool is undefined, once run through the eval, tool = {units:"mm",diameter:6.35,passDepth:4,step:1,rapid:2000,plunge:100,cut:600,zClearance:5,returnHome:true};

openhardwarecoza, 2015-11-16:
You the man @andrewhodel !   Sorted and working!  https://github.com/openhardwarecoza/LaserWeb/commit/75e10ba0df23604f658d4a5f72aedcb00a403cd4

I need to do a tonne of cleanup of console.log statements i used to try to see whats going on. 
Also need to change the default tool etc. But thats easy.  Closing this issue, the remaining things are simple enhancements.

Lesson learned. eval executes the JS :|  - i should get a JS book for xmas and read it over the holidays...

closed 2015-11-16 by openhardwarecoza