LaserWeb/deprecated-LaserWeb1 #5: Add Smoothieboard support

openhardwarecoza, 2015-11-10:
@quillford requested smoothie support in https://plus.google.com/u/0/+PetervanderWalt/posts/H331WFvtDLK

openhardwarecoza, 2015-11-10:
@arthurwolf sponsored me a Smoothieboard to play with

https://plus.google.com/+PetervanderWalt/posts/1wMSLU6RcrW

[image]

quillford, 2015-11-10:
I may be upgrading to a smoothie soon so I could help with this once I get it installed.

openhardwarecoza, 2015-11-10:
Hooked it up just to my PC to spy on the serial in the meantime, just to see what we need to do to support it.

I ran the commands I use to check the status feedback in LaserWeb

>>>M119
SENDING:M119
min_x:1 min_y:1 min_z:1 max_x:1 max_y:1 max_z:1

>>>M114
SENDING:M114
ok C: X:0.000 Y:0.000 Z:0.000 A:0.000 B:0.000 C:0.000  E:0.000

>>>M501
SENDING:M501
File not found: /sd/config-override

>>Triggering a limit switch
Limit switch min_x was hit - reset or M999 required
!!

Comparatively, there are differences:
Command: M114
Smoothie: ok C: X:0.000 Y:0.000 Z:0.000 A:0.000 B:0.000 C:0.000  E:0.000
Marlin:       X:0.00Y:0.00Z:0.00E:0.00 Count X: 0.00Y:0.00Z:0.00

Command: M119
Smoothie: min_x:1 min_y:1 min_z:1 max_x:1 max_y:1 max_z:1
Marlin:       Reporting endstop status
x_min: open
x_max: open
y_min: open
y_max: open
z_min: open
z_max: open

Command: M501
Smoothie: File not found: /sd/config-override
Marlin:       echo:Hardcoded Default Settings Loaded
echo:Steps per unit:
echo:  M92 X53.50 Y53.50 Z3260.00 E5000.00
echo:Maximum feedrates (mm/s):
echo:  M203 X5000.00 Y5000.00 Z5000.00 E5000.00
echo:Maximum Acceleration (mm/s2):
echo:  M201 X9000 Y9000 Z9000 E9000
echo:Acceleration: S=acceleration, T=retract acceleration
echo:  M204 S8000.00 T8000.00
echo:Advanced variables: S=Min feedrate (mm/s), T=Min travel feedrate (mm/s), B=minimum segment time (ms), X=maximum XY jerk (mm/s),  Z=maximum Z jerk (mm/s),  E=maximum E jerk (mm/s)
echo:  M205 S0.00 T0.00 B1 X20.00 Z20.00 E5.00
echo:Home offset (mm):
echo:  M206 X0.00 Y0.00 Z0.00
echo:Laser lifetime usage:
echo: Hours: 0
 Minutes: 0
echo:PID settings:
echo:   M301 P22.20 I1.08 D114.00

Command: Endstop being hit mid-job
Smoothie: Limit switch min_x was hit - reset or M999 required
!! // this makes it stop too, on marlin it continues...
Marlin:       echo:endstops hit:  Y:154.93

So we are going to need two parsers and a config option to switch between them. We can automate the detection:
>>>M115

Marlin:
SENDING:M115
FIRMWARE_NAME:Marlin V1; Sprinter/grbl mashup for gen6 FIRMWARE_URL:http://www.mendel-parts.com PROTOCOL_VERSION:1.0 MACHINE_TYPE:FreeBurn2 EXTRUDER_COUNT:1 (we can parse out firmware name

Smoothie: no response from M115..    http://smoothieware.org/supported-g-codes

quillford, 2015-11-10:
Perhaps it is better to have this as some sort of Mousse module in terms of smoothie support. You can find Mousse [here](https://github.com/quillford/mousse) - that is just my fork but it is the most (and probably only) functional version.

openhardwarecoza, 2015-12-15:
Started adding these in (:

https://github.com/openhardwarecoza/LaserWeb/commit/501c746e9a482e8165c34e36dc69aa1b51cfe8d5

openhardwarecoza, 2015-12-21:
I am closing this Issue, since the scope is now expanding to include Smoothie, Marlin, LasaurGrbl and TinyG!

Commits like https://github.com/openhardwarecoza/LaserWeb/commit/e7ebbcf30e77a20b268ae08f79a4a264d27e9d19 https://github.com/openhardwarecoza/LaserWeb/commit/501c746e9a482e8165c34e36dc69aa1b51cfe8d5 https://github.com/openhardwarecoza/LaserWeb etc form part of this, and Smoothie pretty much works flawlessly now.

openhardwarecoza, 2015-12-21:
As requested by +Tiaan Steyn, Laserweb now also support LasaurGrbl for your Lasersaur users

https://github.com/openhardwarecoza/LaserWeb/commit/e7ebbcf30e77a20b268ae08f79a4a264d27e9d19

- Upon opening a port, we ask it to report its firmware type: https://github.com/openhardwarecoza/LaserWeb/commit/e7ebbcf30e77a20b268ae08f79a4a264d27e9d19#diff-de05965be7003198949126ccae9e71deR101

- Code then autodetects whether we are running Marlin, Smoothie or LasaurGrbl (Note - default Lasaurgrbl runs at 57600 baud, so please adjust according to your setup in config.js before running) https://github.com/openhardwarecoza/LaserWeb/commit/e7ebbcf30e77a20b268ae08f79a4a264d27e9d19#diff-78c12f5adc1848d13b1c6f07055d996eR138  and sets up the queryloop to ask it once a second for position data

-  Changed the Queue management to also support LasaurGrbl (which doesnt send "ok" as acknowlegdement)  https://github.com/openhardwarecoza/LaserWeb/commit/e7ebbcf30e77a20b268ae08f79a4a264d27e9d19#diff-78c12f5adc1848d13b1c6f07055d996eR227

- Display the firmware version on a button on the top navbar - also serves as a definate "Are we connected?" indicator

Todo: 
- Add in the feedback to detect chiller, door, limit hit on the Lasaurgrbl and report in the UI
- Changes to the jog widget (Lasaurgrbl uses G30 to home, Marlin and Smoothie uses G28.  Since we can detect and report the firmware now, its just a case of a little if/else to add
-  Add buttons for Air/Aux enable (lasaurgrbl) - they can be hidden, and when we detect that the  firmware string contains the word "LasaurGrbl" we display them with CSS (see why I added the detection first (; - needed for all these enhancements)

﻿

closed 2015-12-21 by openhardwarecoza

openhardwarecoza, 2015-12-21:
Done with the todo-list for LaserSaur - see https://github.com/openhardwarecoza/LaserWeb/commit/6bb315de5a44177f85f72f1a206dc22414d27b6a

Pretty clever code there, detects what firmware is running and adapts UI accordinly (hides air assist if using something other than lasaurgrbl, and hides the m94, g92 buttons when running lasaurapp.   ASlso changes the gcode sent for homing (g30 for lasaur, g28 for others)

https://github.com/openhardwarecoza/LaserWeb/commit/6bb315de5a44177f85f72f1a206dc22414d27b6a