LaserWeb/deprecated-LaserWeb1 #7: Be able to change settings from GUI

quillford, 2015-11-10:


openhardwarecoza, 2015-11-19:
Right,  so starting to think of tackling this one. 

I am thinking Nosql would be quickest,  easiest.  Config data is nice and flat.

andrewhodel, 2015-11-19:
It's already in reprapweb as a json string saved and loaded to a file. 

That's with many options and as it is in memory it is instant, dbs are overrated for control apps. Dbs are for data storage and even then I'd argue to store to files and use the same type code as an index (that's what a db is). 

You can't count on dbs for long. Before long the requirements list will grow unapproachable. 

There are simple in memory or file dbs as well but is it really that hard to just read and write json?

openhardwarecoza, 2015-11-19:
Aha, then theres even example code for me to work from.  SlOpts?   let me get working on that

openhardwarecoza, 2015-11-19:
https://github.com/andrewhodel/reprapweb/blob/master/server.js#L274 (:

andrewhodel, 2015-11-19:
It's just JavaScript JSON.parse() and Object.stringify(). Object is or contains JSON.

You can prototype any functions to anything.

Cinezaster, 2015-12-24:
https://www.npmjs.com/package/node-persist
[x] Is persistent
[x] writes to a json file
[x] uses localStorage API style

Never used this but looks promising : https://github.com/louischatriot/nedb and works with electron If you ever want to build it as an application.
This can also store files, nice feature if you want to keep the gcode, svg, dxf files the user uploads/loads.

andrewhodel, 2015-12-24:
https://github.com/andrewhodel/reprapweb/blob/master/server.js#L274

From line 274 to 321 does the same thing in just 47 lines of code rather than an entire module you have to trust in the future.

openhardwarecoza, 2016-01-21:
I kicked off work on this!  Too many new issues starting to depend on having this in place:
-- User customisable laser on, laser off, and Sxxx value configs
-- Custom buttons
-- Git pulls not working when you update the ui.js with new bed size

JSON:
[image]

cojarbi, 2016-01-22:
Would be useful to be able to define the name of the file before downloading the gcode

openhardwarecoza, 2016-01-28:
[image]

Status update:

(and dev guide)

1.  The available options are built up from https://github.com/openhardwarecoza/LaserWeb/blob/master/slBaseOpts.js
2.  It gets saved to 'machines' in root of laserweb as json,  loaded on startup, and saved, deleted, edited - by https://github.com/openhardwarecoza/LaserWeb/blob/master/server.js#L531 down to https://github.com/openhardwarecoza/LaserWeb/blob/master/server.js#L627
3.  They get pushed to the frontend, and handled by https://github.com/openhardwarecoza/LaserWeb/blob/master/i/main.js#L150 down to https://github.com/openhardwarecoza/LaserWeb/blob/master/i/main.js#L298

What I am not doing yet is to use https://github.com/openhardwarecoza/LaserWeb/blob/master/i/main.js#L413 to set the variables we need to use elsewhere (start, stop gcode, laser on off and power commands, bed size, etc)

openhardwarecoza, 2016-04-08:
Note that this is WORKING in LaserWeb2 and since LaserWeb2 will soon replace LaserWeb1,  if you need this functionality please checkout LaserWeb2 instead

closed 2016-05-11 by openhardwarecoza