LaserWeb/deprecated-LaserWeb1 #9: Won't start on OS X

quillford, 2015-11-11:
I'm getting an error because LaserWeb tries to initially connect to the bluetooth serial port on os x. This is when I don't have the laser plugged in. 

```
MacBook-Pro:LaserWeb quillford$ node server.js
Got error: connect ECONNREFUSED not enabling webcam
events.js:85
      throw er; // Unhandled 'error' event
            ^
Error: Cannot open /dev/cu.Bluetooth-Modem
    at Error (native)
```

openhardwarecoza, 2015-11-16:
I dont own a mac to test on, but I think this belongs upstream, log an issue on https://github.com/voodootikigod/node-serialport

quillford, 2015-11-16:
I think it's because LaserWeb tries to automatically connect to the available ports. Having the user select the port would probably fix the issue.

openhardwarecoza, 2015-11-16:
If thats the case, it would affect GrblWeb and ReprapWeb too.  @andrewhodel ?

andrewhodel, 2015-11-16:
Most people are using the rpi images I'd imagine, if they are building it from source to run on os x surely they can just type in the port name they want to use in the code.  Or just use the tty as a file like it's supposed to be.... ehrm... windows.

closed 2015-11-18 by openhardwarecoza