LaserWeb/deprecated-LaserWeb1 #14: Unable to start - throw er; // Unhandled 'error' event

zhivko, 2015-12-20:
I installed like per Readme.md and after installing all modulest and starting LaserWeb, it results in:

kz@klemenHp:/mnt/Extra10Gb/git/LaserWeb$ nodejs server.js
Got error: connect ECONNREFUSED not enabling webcam
connected to /dev/ttyS0 at 115200
connected to /dev/ttyS1 at 115200
connected to /dev/ttyS10 at 115200
connected to /dev/ttyS11 at 115200
connected to /dev/ttyS12 at 115200
connected to /dev/ttyS13 at 115200
connected to /dev/ttyS14 at 115200
connected to /dev/ttyS15 at 115200
connected to /dev/ttyS16 at 115200
connected to /dev/ttyS17 at 115200
connected to /dev/ttyS18 at 115200
connected to /dev/ttyS19 at 115200
connected to /dev/ttyS2 at 115200
connected to /dev/ttyS20 at 115200
connected to /dev/ttyS21 at 115200
connected to /dev/ttyS22 at 115200
connected to /dev/ttyS23 at 115200
connected to /dev/ttyS24 at 115200
connected to /dev/ttyS27 at 115200
connected to /dev/ttyS28 at 115200
connected to /dev/ttyS29 at 115200
connected to /dev/ttyS3 at 115200
connected to /dev/ttyS30 at 115200
connected to /dev/ttyS31 at 115200
connected to /dev/ttyS4 at 115200
connected to /dev/ttyS5 at 115200
connected to /dev/ttyS6 at 115200
connected to /dev/ttyS7 at 115200
connected to /dev/ttyS8 at 115200
connected to /dev/ttyS9 at 115200
connected to /dev/ttyUSB0 at 115200
connected to /dev/ttyS25 at 115200
connected to /dev/ttyS26 at 115200
connected to /dev/ttyACM0 at 115200

events.js:72
        throw er; // Unhandled 'error' event

here is pastebin of whole procedure:
http://pastebin.com/Ui1yjtSv

openhardwarecoza, 2015-12-21:
Running 'sudo rm /dev/ttyS* before launching should take care of it.

In linux, if you have a hardware serial port, it loads the serial port kernel module that creates all these ttyS* devices, however THEY DONT all EXIST! Nor are we interested in interacting with them

If you have a hardware serial port on your motherboard, which you never use, disable it in your BIOS, then these devices will no longer appear

closed 2015-12-21 by openhardwarecoza

openhardwarecoza, 2015-12-25:
HI @zhivko , I added some code to handle this:

https://github.com/openhardwarecoza/LaserWeb/commit/eda08a55019d3f4f84268b77baa3b06aeb9bc2e7

in response to https://github.com/openhardwarecoza/LaserWeb/issues/14  

Now you no longer need to rm the ttyS ports...

It will gracefully report that it could not write to them and then skip (without crashing)

zhivko, 2015-12-25:
I could use laserweb after removing tty as you suggested - thanks!

openhardwarecoza, 2015-12-26:
@zhivko  Now also update your install (run git pull while in the LaserWeb directory)

I update almost daily, so make sure you do so you have latest features / bugfixes

Just look at how much i have done,  by looking at the commit log https://github.com/openhardwarecoza/LaserWeb/commits/master since your download (: