LaserWeb/deprecated-LaserWeb1 #15: Application crashes if no device connected

danawoodman, 2015-12-23:
First off, great project!

I'm unable to start LaserWeb without a machine connected with the following error. Seems to be looking for any device and throws if no device can be found. Would be nice to catch this exception and let the user know somehow

```
> reprapweb@ start /Users/dana/code/LaserWeb
> node server.js

Got error: connect ECONNREFUSED not enabling webcam
connected to /dev/cu.Bluetooth-Incoming-Port at 115200
events.js:85
      throw er; // Unhandled 'error' event
            ^
Error: Cannot open /dev/cu.JabraREVOa400-SPPDev
    at Error (native)

npm ERR! Darwin 15.0.0
npm ERR! argv "/Users/dana/.nvm/versions/node/v0.12.7/bin/node" "/Users/dana/.nvm/versions/node/v0.12.7/bin/npm" "start"
npm ERR! node v0.12.7
npm ERR! npm  v2.11.3
npm ERR! code ELIFECYCLE
npm ERR! reprapweb@ start: `node server.js`
npm ERR! Exit status 1
npm ERR!
npm ERR! Failed at the reprapweb@ start script 'node server.js'.
npm ERR! This is most likely a problem with the reprapweb package,
npm ERR! not with npm itself.
npm ERR! Tell the author that this fails on your system:
npm ERR!     node server.js
npm ERR! You can get their info via:
npm ERR!     npm owner ls reprapweb
npm ERR! There is likely additional logging output above.

npm ERR! Please include the following file with any support request:
npm ERR!     /Users/dana/code/LaserWeb/npm-debug.log
```

openhardwarecoza, 2015-12-24:
Thanks for the error handling code!

closed 2015-12-24 by openhardwarecoza

openhardwarecoza, 2015-12-25:
I added some further improvements:  

https://github.com/openhardwarecoza/LaserWeb/commit/eda08a55019d3f4f84268b77baa3b06aeb9bc2e7

to handle https://github.com/openhardwarecoza/LaserWeb/issues/14