LaserWeb/deprecated-LaserWeb1 #21: Improve serial bridge

openhardwarecoza, 2015-12-24:
Open to options here.  

SerialPort + windows users suffer,  are forced to use older versions of node or compile SerialPort native module from source.    

No,  we cant say we dont support windows (lasercutter people are graphics people so i expect mac+win to be the majority 

Win no 1 would be a precompiled executable doing everything server.js does.  Cross platform,  too

Or lets explore chrome app / serial api etc

openhardwarecoza, 2015-12-24:
@danawoodman @andrewhodel @quillford

Cinezaster, 2015-12-24:
Make an [Electron app](http://electron.atom.io/) from it.
Someone made this work https://gist.github.com/jedthehumanoid/a7f8278e0a37d259adca for windows.

I only have experience on osx

openhardwarecoza, 2015-12-24:
That looks ideal in quick overview! Same code we already have, just built as a desktop app...

Investigating!

danawoodman, 2015-12-24:
Electron would be sweet! Having everything bundled together and bootable by just launching the app would be a win for usability. My makerspace users aren't all technical so this would be a big plus in my book. 

I know Chilipeppr uses this https://github.com/johnlauer/serial-port-json-server. That or the Chrome serial API https://developer.chrome.com/apps/serial

Maybe it can work? It is cross platform.

andrewhodel, 2015-12-24:
I have a few days with nothing to do here, I'll try making a rrw and gw version that uses the chromium serial api and see what happens.  Then you can test it on Windows.

I'll share the Javascript by updating those projects then you can merge if it all works.

I've been working with RPI and tty for a navigation project and that is working great with very little code.

Before running the app, this configures the tty:

```
stty -F /dev/ttyUSB0 4800 -brkint -icrnl -imaxbel -opost -isig -icanon -iexten -echo -echoe -echok -echoctl -echoke

cat /dev/ttyUSB0 | node app.js
```

Code for app.js

```
var readline = require('readline');
var rl = readline.createInterface({
  input: process.stdin,
  output: process.stdout,
  terminal: false
});

rl.on('line', function(line){
    console.log(line);
});
```

Nonetheless I like the idea of the chrome serial api, it seems very cool.

andrewhodel, 2015-12-24:
Electron looks cool as well, I will also check that out and thanks for sharing.

openhardwarecoza, 2015-12-25:
Thanks @andrewhodel!  

Yip unfortunately actually,  this project has outgrown the pi image concept (unfortunate because thats really the easiest to support for distribution -  as you can also control the environment)

openhardwarecoza, 2016-01-21:
I'm going to close this one for now.  With good documentation as contributed by the community lately. it's been a while since someone had issues installing.  Keep it in back of mind, will still accept Pull requests (;

openhardwarecoza, 2016-01-30:
Reopening... We have some issues where Node Serialport doesnt send all the data to the device

funinthefalls, 2016-01-30:
Have you taken a look at nodewebkit?

openhardwarecoza, 2016-01-30:
Wont help,  the issues is in Node SerialPort so alternatives boil down to more than just install.

reconbot, 2016-04-06:
Hi,
We've got a new beta version ready for testing that fixes a lot of issues on windows. This includes throwing some errors that might have been lost that helps explain bad situations. I don't know too much about your project but I'd love to help debug. You can learn more about `serialport@2.0.7-beta4` here  https://github.com/voodootikigod/node-serialport/issues/733

openhardwarecoza, 2016-04-07:
Awesome,  we are busy moving away from node-serialport though (see github.com/openhardwarecoza/LaserWeb2  )  but hey,  any existing LaserWeb1 users,  feel free to give this a try (:

reconbot, 2016-04-07:
I'm sorry the project dove you away! Besides fixing bugs and improving the api. 

We're approaching parity with chrome's own serial object. (we'll actually have more features!) We've got a project called `browser-serialport` which wants to maintain a compatible api that uses the chrome's serialport as it's backend. That way you could switch to an electron app and keep the same code.

https://www.npmjs.com/package/browser-serialport

openhardwarecoza, 2016-04-07:
Well i am always happy to go back and forth (: -  but my time is limited
and contributors too.   If you can help out,  i would love assistance with
playing with that.  (i am self taught, so excuse the mess which is the
code)

reconbot, 2016-04-07:
Time always is. I wanted to make sure you knew what we were up to, as maybe
it might save you some in the future. =)

openhardwarecoza, 2016-05-11:
Serial Port JSON server under LaserWeb2 works OK so closing this for now

closed 2016-05-11 by openhardwarecoza