LaserWeb/deprecated-LaserWeb1 #23: Missing M-Code in millcrum mc.js

oxydum, 2015-12-24:
To switch the laser diode on, 
  		this.gcode += 'M03'+'\n';

To switch the laser diode off, 
		this.gcode += 'M05'+'\n';

To end the program,
		this.gcode += 'M02\n';

That's working for me. ;)

openhardwarecoza, 2015-12-24:
Can you do a pull request with the changes please (:

@Cinezaster , we can add your https://github.com/openhardwarecoza/LaserWeb/issues/24 to this as well:

In mc.js from https://github.com/openhardwarecoza/LaserWeb/blob/master/i/millcrum/inc/mc.js#L746  down

Add a field for cutting intensity to index.html https://github.com/openhardwarecoza/LaserWeb/blob/master/i/index.html#L644

Bonus points for anyone who adapts the DXF/SVG importers to generate a few entries in table format on that modal, that allows custom feed/power per path (;

Extra bonus points if we then add the cut on path, inside, outside - then we can use this with CNC too

openhardwarecoza, 2015-12-24:
Wife is having me do xmas'sy stuff so no coding for me today ):

Cinezaster, 2015-12-24:
I'm stuck on xmas duty as well and leaving tomorrow till 4th of January for some skiing.
I already have experience with SVG(/DXF) and turning them into Gcode.
https://github.com/Cinezaster/svgPathToRoboCode This was build for controlling and operations a scara robot that would draw lines of light in the air. It communicated over serial over bluetooth with the robot arm. But I discovered recently that the code was also use full and optimized for generating gcode. 
A new dedicated attempt for coverting svg's to gcode for lasers, a bit the same direction of LaserWeb. https://github.com/Cinezaster/svg2gcode4lasers. Got to much work on my hand lately. So on hold. Would be glad to help out.

openhardwarecoza, 2015-12-24:
@andrewhodel did all the grunt work for the svg/dxf already (written from scratch) so happy with that.  I also have an idea on how to do what I want to do above (multi paths, even multi color) so lets see if I get space before 4 Jan.  I am not taking leave this year - working at the office throughout

andrewhodel, 2015-12-24:
DXF - https://github.com/andrewhodel/millcrum/blob/master/inc/dxf.js

SVG - https://github.com/andrewhodel/millcrum/blob/master/inc/svg.js

They work quite well and are pure JS with no nonsense all from scratch.  Really easy to include in other projects.

openhardwarecoza, 2015-12-25:
I can vouch for the really easy part.   I a a hardware guy who self taught myself JS from May this year,  and I could integrate them into laserweb in about two days (javascript Eval() part of millcrum to gcode  was the most confusing part,  as i taught myself and had no idea what eval does lol)

openhardwarecoza, 2015-12-25:
@oxydum , I did a remote pull of your patch:

https://github.com/openhardwarecoza/LaserWeb/commit/0088203ad31f6a5f57d439af97be0b3f1d5724a9

Thanks! Much better than M3/M5 for each segment I had 

I still like the marlin/smoothie/lasaurgrbl way of using G0 (MOVE) vs G1 (CUT) (;

openhardwarecoza, 2015-12-25:
Ok, I started working on the improved importer UI, for now DXF only:

https://github.com/openhardwarecoza/LaserWeb/commit/e4ff991e3ea8bf4ec19507a38c4a875ba26e2411
https://github.com/openhardwarecoza/LaserWeb/commit/76df0b5f6cbe6b1f13e8be64cb93b2261a9f8ec2
and
https://github.com/openhardwarecoza/LaserWeb/commit/5762e95d1aad20cbf0746be4bbfaaf93eada37c5

[image]

openhardwarecoza, 2015-12-25:
Its fully functional, by the way

But that UI needs a cleanup! Drastically (currently that's the worst looking part of LaserWeb (;   hehe)

And my hack of mc.js might be questionable - cross referenced to https://github.com/andrewhodel/millcrum/issues/14

openhardwarecoza, 2015-12-25:
[image]

Cleaned up the UI, and also added this functionality to SVG importer! (:

Looks like we'll stick with hacked mc.js (based on response in https://github.com/andrewhodel/millcrum/issues/14

Awaiting your comments!

closed 2015-12-27 by openhardwarecoza