LaserWeb/deprecated-LaserWeb1 #28: Hangs on connecting to Laser Cutter/Engraver serial port.

funinthefalls, 2015-12-25:


openhardwarecoza, 2015-12-25:
Can you give me more details on your setup? 

Os
Board 
Firmware? 

Etc

funinthefalls, 2015-12-25:
OS: Windows 10
Board Ramps 1.4/Arduino Mega2560 with RepRap full graphic LCD
Firmware: most recent build of Turnkey Tyranny's Marlin
GCode creation: Inkscape---->Turnkey Tyranny's Laser Exporter (for both vector and raster) --->Pronterface = Success
GCode creation: Inkscape---->Turnkey Tyranny's Laser Exporter (for both vector and raster) --->Laserweb = Failure to connect

This is running on a 6 core AMD phenom with 16GB Ram, 1 GB video card

openhardwarecoza, 2015-12-26:
Screenshot of node terminal showing the hang?

openhardwarecoza, 2015-12-27:
Other mental checks:

NB: do a 'git pull' in the LaserWeb directory to make sure you get the latest code

1.  Make sure board is plugged in before starting laserweb
2.  Make sure pronterface is NOT still open (else it would hold on to the port)
3.  You should see the terminal do something like this
[image]

Note that if you get that far and the fact that nothing else prints on the terminal, is what you consider a HANG, it means you havent read the readme (https://github.com/openhardwarecoza/LaserWeb/blob/master/README.md#access)

After starting laserweb, you need to open a web browser to http://127.0.0.1:8000 (or http://ip-of-your-server:8000)

[image]

closed 2015-12-27 by openhardwarecoza

openhardwarecoza, 2015-12-27:
I have updated server.js to print you a message with instructions on where to go next:
[image]

And I have also updated the README:  See  https://github.com/openhardwarecoza/LaserWeb#access-instructions