LaserWeb/deprecated-LaserWeb1 #29: New error when trying to start node

funinthefalls, 2015-12-26:
This is what I get when I try to load node server:

Microsoft Windows [Version 10.0.10586]
(c) 2015 Microsoft Corporation. All rights reserved.

C:\Users\Anthony>cd\

C:\>cd laserweb

C:\LaserWeb>node server
module.js:338
    throw err;
          ^
Error: Cannot find module 'dotenv'
    at Function.Module._resolveFilename (module.js:336:15)
    at Function.Module._load (module.js:278:25)
    at Module.require (module.js:365:17)
    at require (module.js:384:17)
    at Object.<anonymous> (C:\LaserWeb\config.js:1:63)
    at Module._compile (module.js:460:26)
    at Object.Module._extensions..js (module.js:478:10)
    at Module.load (module.js:355:32)
    at Function.Module._load (module.js:310:12)
    at Module.require (module.js:365:17)

C:\LaserWeb>

openhardwarecoza, 2015-12-26:
Looks like you skipped the 'npm install'

funinthefalls, 2015-12-26:
Yup, dumb move on my part. Thanks for the help

closed 2015-12-26 by funinthefalls