LaserWeb/deprecated-LaserWeb1 #30: Connected but button says not connected

cojarbi, 2015-12-27:
Im able to connect to the azsmz board and run a gcode but the upper right button still show not connected
[image]

openhardwarecoza, 2015-12-27:
In https://github.com/openhardwarecoza/LaserWeb/blob/master/server.js#L177 to line 262, we check what response the board sends,  and if it matches one of those,  it shows connected in the ui.  

Your board must be printing something different than a real smoothieboard.  

If you could help me find what it prints:
1.  Edit server.js
2.  Take out the two // before console.log on line 
https://github.com/openhardwarecoza/LaserWeb/blob/master/server.js#L174
3. With the board plugged in,  start server.js
4.  In the terminal it should print out near the top,  before starting to print the position statements, a few lines,  one of which is the output from the 'version'  command on smoothieware.   Copy paste that text here,  and i will add a extra check to the code for azmz (i dont have one one hand.  But with the console.log enabled,  thats how i add boards: check what they say,  and add code to handle it

openhardwarecoza, 2015-12-27:
I am taking a wild guess, but after looking at the AZSMZ's wiki page, seems they are using the LPC1768 processor, and a real Smoothieboard uses LPC 1769

So I added a 'or' check for 1769 or 1768 = smoothieware:

See https://github.com/openhardwarecoza/LaserWeb/commit/3d5d17e28076207bb30a63e942fe5206e458b238

So, do a 'git pull' inside your LaserWeb directory, fire up server.js and let me know if it works.

If it doesnt, refer to the post above, we'll need to see what your board says then

cojarbi, 2015-12-28:
Yes, now it shows connected. Mind that i did get the following in terminal once i started the server

Missing error handler on `socket`.
TypeError: Cannot read property 'port' of undefined
    at Socket.<anonymous> (/Users/ayahni/LaserWeb/server.js:561:26)
    at Socket.emit (events.js:107:17)
    at Socket.onevent (/Users/ayahni/LaserWeb/node_modules/socket.io/lib/socket.js:330:8)
    at Socket.onpacket (/Users/ayahni/LaserWeb/node_modules/socket.io/lib/socket.js:290:12)
    at Client.ondecoded (/Users/ayahni/LaserWeb/node_modules/socket.io/lib/client.js:193:14)
    at Decoder.Emitter.emit (/Users/ayahni/LaserWeb/node_modules/socket.io/node_modules/socket.io-parser/node_modules/component-emitter/index.js:134:20)
    at Decoder.add (/Users/ayahni/LaserWeb/node_modules/socket.io/node_modules/socket.io-parser/index.js:247:12)
    at Client.ondata (/Users/ayahni/LaserWeb/node_modules/socket.io/lib/client.js:175:18)
    at Socket.emit (events.js:107:17)
    at Socket.onPacket (/Users/ayahni/LaserWeb/node_modules/socket.io/node_modules/engine.io/lib/socket.js:99:14)

openhardwarecoza, 2015-12-28:
Cosmetic error - sometimes pops up. Did it work despite this (i.e. could you jog machine, etc?)

cojarbi, 2015-12-28:
Yes I did work correctly. I did not loaded any file just used the jog

openhardwarecoza, 2015-12-28:
Cool.  Here's some working GCODE and DXFs to play with.  At the moment SVG is a bit broken (see #23  )

[samples.zip](https://github.com/openhardwarecoza/LaserWeb/files/73167/samples.zip)

closed 2015-12-28 by openhardwarecoza