LaserWeb/deprecated-LaserWeb1 #31: Error disconnecting board

cojarbi, 2015-12-30:
Im getting the following error when the board is disconnected from the computer.

events.js:85
      throw er; // Unhandled 'error' event
            ^
Error: Serialport not open.
    at SerialPortFactory.SerialPort.write (/Users/ayahni/LaserWeb/node_modules/serialport/serialport.js:288:17)
    at null.<anonymous> (/Users/ayahni/LaserWeb/server.js:261:20)
    at wrapper [as _onTimeout] (timers.js:274:14)
    at Timer.listOnTimeout (timers.js:119:15)

quillford, 2015-12-30:
I haven't had time to use LaserWeb recently, so I'm not sure if there is one but I think a connect/disconnect button may fix this (assuming users use it).

openhardwarecoza, 2015-12-30:
We automatically connect to all ports on startup.   

That way server.js can handle background tasks on other ports while you work on another port (suppose you have two or three machines.  Kickoff a job againt /dev/ttyACM0 (runtime 15 min).   Server.js keeps sending gcode to acm0.  In ui,  i can use the dropdown to activate acm1.   Jog and prep,  start another job.  

Laserweb's core in that sense will soon handle cnc,  laser and 3d printer (so technically you can have just one webapp,  drive all machines connected to that host.  

What we really need to fix that disconnect crash is some error handling that on disconnect,  scans SerialPort again,  and emits the list of ports,  again.   Same for when a new device gets plugged in (or a refresh button) 

But keep the functionality of connecting to everything

openhardwarecoza, 2015-12-30:
Basically  do https://github.com/openhardwarecoza/LaserWeb/blob/master/server.js#L107 on startup,  again when a board disconnects,  and again when user clicks a refresh ports button?

openhardwarecoza, 2016-01-01:
As a side note, I run my server.js with forever 

npm install -g forever
forever start server.js

Forever detects when the process dies and starts it again for you. Fine for single machine setups (if you disconnect running job is anyway ruined) - not OK if you have 2+ machines on same instance of course...

closed 2016-01-10 by openhardwarecoza