LaserWeb/deprecated-LaserWeb1 #32: Adding Raster Support

openhardwarecoza, 2015-12-31:
I am opening an issue on my own project - since it allows me a bit of a work log functionality.

Building raster is a bit unlike any of the other things I have done with laserweb, because there is very little to build from:
With the core serialport/nodejs code in Laserweb, I based off the work from GrblWeb
With the UI work, I have been using bootstrap for years
With 3D GCode viewer and three.js - I based my work off the work by http://joewalnes.com/
For the parametric apps, opejscad.org has been a great help as have been the scripts from their relevant original authors (as mentioned in each's header). 

Sure it was still  LOT of work adapting their various code to my project.  Sometimes I think adapting is more work than rewriting.  In fact thats what LaserWeb owes its existence too - I got tired of adapting other hosts - so I decided to write my own...

So, lets begin with Raster!

This morning I woke up and checked Google Plus, to find one of my followers who posted a quick video running some raster gcode he found online, via LaserWeb:

See post here: https://plus.google.com/+ArielYahni/posts/3GMYsfp2AVN

That was all the inspiration I needed to decide, I am going out of 2015 with working Raster support!

So, a few weeks now I have been doing the prep work, and found something called PaperScript.  Paperscript is an extension of Javascript, with Paper.js:
Paper.js has a Rasterise function that scans a bitmap and puts each pixel into an array.  
Probably took me an hour to get the basics working:  

[image]

Here all we do is scan in the image using paper.js's raster and then use that data to draw on a HTML5 canvas, a series of SVG paths (circles), in the same coordinates where it found a dot in the bitmap.  Then we simply scale the size of the dot according to the percentage of Gray in that dot (between 0 and 1 where 1 = 100% = black) 

Thats of course a perfect aproximation of what a laser would do if we were to set the output PWM according to the Gray level of that pixel. Or in other words, proper grayscale raster engraving - long the holy grail of rastering.   The higher the laser power on that dot, the darker/larger that dot will appear.  So in theory the simulation I have done so far then should be quantitively similar to real world results

Next came adding GCODE generation.  Damn that was just too easy! For each line I do a G0 Yx move, and then we raster along the X axis in G1 Xxx Yxx Sxx commands.  Where Sxx can be a value based on the gray % of that dot.

This brought me to:

[image]

Thats pretty close to a working implementation then!

So what's missing:

1.  I need to add some form control to the UI to allow you to select things like feedrate, laser min/max range (for engraving you dont want 0-100% - a black dot may only need 20% power and a 50% gray dot needs 10% power, for example) - so simple math to map 0-1 in gray values - to x%-y% in gcode. 

2.  Have to resize the canvas on loading the image (canvas has to be same size as image * gridSize

3.  Add some calculation of real world engraving size.  Also make that scaling (gridSize) configurable
Instead of 

``` posx = x /10```
``` posy = y/10```

should be doing

``` posx = x /scaleFactor```
``` posy = y/scaleFactor ```

Then finally integrating back into Laserweb's UI.

Once in the UI we can just diable the 3D viewer and display the rastered canvas instead (raster gcode on a 3D Gcode viewer will only show a solid black square since its all G1 moves anyway)

openhardwarecoza, 2015-12-31:
Commited 

https://github.com/openhardwarecoza/LaserWeb/commit/04db55b43b5e8a3f7445ecd6974f51a3eb25fabb

:( - sadly last night I was also doing some renaming of folders, cleaning up code etc and forgot to commit that seperately, so this commit also includes some openjscad related commits.

If you just wanted to see the raster work, look at the commits related to files in /i/raster only!

  For now ALL the

openhardwarecoza, 2015-12-31:
Mostly https://github.com/openhardwarecoza/LaserWeb/blob/04db55b43b5e8a3f7445ecd6974f51a3eb25fabb/i/raster/index.html should be of interest for now...

martinxyz, 2015-12-31:
Author of Lasersaur pulseraster firmware/branch here. I'm interested in how this goes, I'll be watching this.

The G-code processing (and even transmitting over a serial port) is going to be a serious performance bottleneck with this approach, especially with Marlin-style firmware, which has a large per-line-segment overhead. But this sure beats having no raster support at all. With the Lasersaur, I'm aiming for 8000mm/min rastering with a 100W laser at 8% duty and 5 pixels per mm (black or white). Grayscale image are dithered first to binary using Floyd-Steinberg (which doesn't seem to hurt the quality, actually I'm starting to think it even improves it).

For large black letters or black shapes (instead of photos) I've recently seen someone code it up in a LasaurApp fork as gcode lines with variable length. This should perform well even with a firmware that has no raster support. Here is the implementation (Python): https://github.com/gregov/LasaurApp/blob/master/backend/filereaders/svg_tag_reader.py#L230

If there is a GUI, I may be interested later in adding support for my new raster protocol for the Lasersaur. But it depends on how things are going elsewhere, so don't hold your breath :-)

openhardwarecoza, 2015-12-31:
Hi Martin
I am actually a moderator on the Lasersaur google group, so well aware of your work.

I used to run my machines on Lasaurapp for a long time (with my own hardware - https://github.com/openhardwarecoza/OPHW-HW-Laser-H4H/tree/master/Electronics%20V3 based of the Driverboard)

With Laserweb, I am worried it would go the path of many other projects - dies because no one uses it - so I might be making a mistake I will regret later, but thats why I welcome all firmwares (:
Grbl gets all those chinese diode laser users in, Lasaurgrbl support means you guys can have a play too, the smoothieware is quickly becoming a favourite, Marlin has been used on hundreds of 40W K40 machine conversions to RAMPS (as unsafe as that is!) - and adding support for all these mean we should get a decent spread of users too (i hope)

My usual dev method is to first get things going in a seperate standalone fashion then add back to my main app.   See my git log for the OpenJSCAD work as an example - so should be in LaserWeb directly by next week or so..

Regarding the different protocols: I already identify the firmware and push that to the GUI:
https://github.com/openhardwarecoza/LaserWeb/blob/master/server.js#L189  and  https://github.com/openhardwarecoza/LaserWeb/blob/master/server.js#L566

ANd in the gui I can then have IF/Else:  for example https://github.com/openhardwarecoza/LaserWeb/blob/master/i/main.js#L860

I can as easilly use that if (firmware.indexOf('Lasaur') == 0) code to have a seperate gcode writer for lasersaur.  I am ANYWAY going to have to do it since there are differences even in the pure gcode methods. (grbl has s0-255, but smoothie has s0-1 (S0.5 in smoothie = S128 in grbl = 50 %) 
For Marlin there is also the Base64 method https://github.com/TurnkeyTyranny/buildlog-lasercutter-marlin/blob/master/Marlin/Marlin_main.cpp#L1070

So while my initial implementation will be just plain Gcode, the framework is there, ready to go, for anyone to add a custom writer.

Then - things like optimising the dithering etc, for now, is above my skill level - i am relying on what paper.js can do for me here (which is quite powerful, I can set spacing between dots to sample, etc as well) - but I could REALLY use help with that.

So basically in the end we could have that if code generate 
Gcode for smoothie and grbl (smoothie we save to SD and stream from there - so serial isnt an issue)
Base 64 for Marlin
and your protocol too...  Based on the firmware.indexOf('xxx') code

Lasaurapp itself feels like development has stalled, so I would be happy if any work I do here can help there too!

openhardwarecoza, 2015-12-31:
Oh, and I could really use help from the Lasaurapp guys on getting their excellent SVG support into here ( See   https://github.com/openhardwarecoza/LaserWeb/issues/27 - I had to rip out the library I was using due to the upstream developer dropping support basically...)

openhardwarecoza, 2015-12-31:
Another thing, if I havent mentioned it, is unlike a lot of open source projects, I really do welcome other developers to take ownership of the projects too - I really don't need this to be about me (: - I get my ego stroked enough - so if anyone wants to have push privedges / become a maintainer - please shout!

martinxyz, 2015-12-31:
Thank you for the pointers. It would be a bit more work than just replacing gcode parameters - the new protocol is binary, with a different flow control mechanism: https://github.com/stefanix/LasaurApp/blob/alpha/firmware/src/protocol.c#L21

If LaserWeb stays active and LasaurApp stays dead I will also consider contributing improved SVG support and add support for the new firmware, if my time permits next year. But I'm much more familiar with C and Python than with Javascript. So I watch and see if there remains some level of activity and users here, because this project is still very new, and I wait if maybe some of the web-developer people at our fablab are motivated to invest a lot of time and put something together. Fixing LasaurApp (server-side Python) would be a better match for my current skills, but I also prefer to work on projects that reach more users.

At the moment I think the LaserWeb SVG importer completely ignores colors, and I don't think LasaurApp users will accept a replacement unless they can still use existing SVGs with one color for cut and one for engrave.

Anyway, I guess I should stop abusing this issue for general discussion :-)

openhardwarecoza, 2015-12-31:
Worse,  for now i commented out the svg button completely until we can get #27 fixed.   So i do agree

openhardwarecoza, 2016-01-01:
Ok, so this morning I worked on the optimisation a little:
If there is a few pixels from left to right sharing the same intensity, I concatenate them into one G1 move:

Sample image I used:
[image]

Pulled into LaserWeb:
[image]

and resultant gcode:
[Test Raster.zip](https://github.com/openhardwarecoza/LaserWeb/files/75869/Test.Raster.zip)

I don't have a diode laser nor is Freeburn2 up and running so PLEASE test it on a machine of yours?

openhardwarecoza, 2016-01-01:
Here's the commit for work up to above:

https://github.com/openhardwarecoza/LaserWeb/commit/4ab0a9c7c5fd2ba99536485d88a5e33da8b9ad13

The optimising happens around https://github.com/openhardwarecoza/LaserWeb/commit/4ab0a9c7c5fd2ba99536485d88a5e33da8b9ad13#diff-b65de7a50a1f4dd9293c541edaf44292R125

openhardwarecoza, 2016-01-02:
Ok, so one of my community members ran my test gcode and reported back a few weird issues:

So moving forward:
1. I focussed my thoughs on the 'paths' instead, ignore the gcode generator, I am first intending to draw on the canvas, what I WANT the machine to do (yesterday I was drawing on the cabas what I  was interpreting in the raster and hoping the machine would follow - thats what happens when you code after new years eve...  My logic was flawed.  I should not draw one thing,  and generate gcode for another! I should draw 'paths' and then convert those paths to gcode...

2.  I parse the image and then use that information to draw LINES (no longer dots) from the LAST X/Y POS, to the current XY POS at the LAST PIXELs intensity while counting pixels ++ until a change in intensity is needed...)

https://github.com/openhardwarecoza/LaserWeb/commit/0ac4066f24730ba106c730fdf5c12471b96582d4#diff-b65de7a50a1f4dd9293c541edaf44292R228

3.  The resulting render should represent correctly what the move will be doing (starting where last move ended, and moving to last pixel in that row that shares the same laser power, while drawing it at the power of the laserpower on the first pixel in that group (since they are all the same laserpwr within that group)

[image]

https://github.com/openhardwarecoza/LaserWeb/commit/0ac4066f24730ba106c730fdf5c12471b96582d4#diff-b65de7a50a1f4dd9293c541edaf44292R227 is plaintext of what the gcode should do, so next to fix up the gcode writer

I also extended the UI with a set of sliders (well a two-in-one slider) to constrain the laser.  Since 0-100% black wouldnt be 0-100% power on a laser. It might be mapping to 10-40%  in reality

openhardwarecoza, 2016-01-02:
Gcode writer added again.   Also reversed the Y coordinates (machines work from bottom left  paper.js rasters from top left.)  

Tomorrow i need to add the constraint mapping code.  

And then clean up that UI.  Its a mess

openhardwarecoza, 2016-01-04:
Here's a new test file to try

[rastertest2.gcode.zip](https://github.com/openhardwarecoza/LaserWeb/files/76920/rastertest2.gcode.zip)

openhardwarecoza, 2016-01-04:
Today I did one of the most IMPORTANT parts of this project, I added Sxxx parameter support to the 3D viewer

All the required changes is contained in this one commit: https://github.com/openhardwarecoza/LaserWeb/commit/2f1bb5e9d606d909ce53ba0d5a7c6331cdd9089f

And here's how it looks!

[image]

openhardwarecoza, 2016-01-04:
Here's the file used for the photograph above  

[LargeLady-LaserWeb.gcode.zip](https://github.com/openhardwarecoza/LaserWeb/files/77214/LargeLady-LaserWeb.gcode.zip)

openhardwarecoza, 2016-01-04:
And its integrated into the main UI

[image]

To do:
- add the firmware specific code like m3/m5s for grbl, and the s0-1 for smoothie, s0-255 for the others, etc
- Add the contrain code - that laser power sliders in the screenshot above, just dummy for now :(
- Fix a bug in that the image size only updates when you change the spot size, and doesnt update on load. It works in the standalone but not after integration.

iconexperience, 2016-01-05:
Very nice. I notice there is some heavy banding in your images. To get rid of that you may want to add some good old dithering (<a href="https://en.wikipedia.org/wiki/Floyd%E2%80%93Steinberg_dithering">Flyod-Steinberg dither on Wikipedia</a>) to your process. That should further improve your results.

openhardwarecoza, 2016-01-05:
In practise it comes out perfect, the 'banding' you see is actually the three.js render, depending on zoom.  If you zoom in, it looks better. Download Laserweb, open one of you own images and check (:
See https://www.youtube.com/watch?v=_tbCIdONdZQ

iconexperience, 2016-01-05:
If your results are already good, that's even better. Too bad nobody needs those dithering algos anymore, they are quite clever.

Anyway, your project is great! I hope that I will find more time in future for doing things like this.

openhardwarecoza, 2016-01-05:
Have a look at the code before you judge it (:

If a number of pixels share the same shade: It draws a line (much more efficient than a bunch of dots that a dither would produce): https://github.com/openhardwarecoza/LaserWeb/blob/master/i/index.html#L167

Then maps specific laser PWM inputs to the grayscale value:https://github.com/openhardwarecoza/LaserWeb/blob/master/i/index.html#L181

before generating https://github.com/openhardwarecoza/LaserWeb/blob/master/i/index.html#L219 where  we put a proprtional grayscale value into the Sxxx paramater that the firmware can understand.  

Why, therefore, @iconexperience , the critism and sarcasm? Real grayscale surely outperforms dither

openhardwarecoza, 2016-01-05:
[image]
@iconexperience , view this in fullscreen original size.  See?

iconexperience, 2016-01-05:
My post was neither meant to be critical nor sarcastical. I simply wanted to make a suggestion how the banding (that seemed to exist from the screenshots) could be approached. Of course real grayscale is better than dithering. By saying *"If your results are already good, that's even better."* I meant exactly that, nothing else.

openhardwarecoza, 2016-01-05:
Ahh ok cool.  I hate text. It always fails to carry the emotion,  leading to misunderstandings   (:

martinxyz, 2016-01-06:
@iconexperience Using dithering would cancel out the optimization for rastering with gcode lines. Without firmware raster support it's expected to slow down a lot if you change intensity for every pixel. Be assured that Flyod-Steinberg is still in use, I used it for the Lasersaur raster support. For comparison, here is a macro-photo of how dithered raster pulses look on wood. (The rings are characteristic of the laser tube, the radius depends on the focus.)
[image]

openhardwarecoza, 2016-01-06:
Added a slider for the zoom level (pixel size X pixels = dimensions of final cut:

Also have this draw a bounding rectangle on the 3D viewer that dynamiclly resizes to match while you drag the slider - so you can see how large in relation to the bed - your job is.

[image]

iconexperience, 2016-01-06:
Thanks for all the information. Interesting to see the different rasterization techniques being applied, and I fully understand why using lines is the way to go. You have probably already seen the spiral raster at http://paperjs.org/examples/spiral-raster/ , which is more a gadget, but you maight get some inspiration from that.

openhardwarecoza, 2016-01-06:
@iconexperience that might make a nice 'artsy' effect plugin at some point

openhardwarecoza, 2016-01-07:
ThanktK on GooglePlus suggested adding a 'back stroke' too - to massively decrease job tip (it engraves when going back from right to left too:

Here's the new render. Does a G0 move down the Y axis at the end of each line, iterates through the x coordinates backward, get to the start of the line, goes down the Y, and raster left to right. Repeat

[image]

martinxyz, 2016-01-07:
@openhardwarecoza please consider calling this "bidirectional" instead of "backstroke". I've seen this term in another codebase too, it's not my invention.

openhardwarecoza, 2016-01-07:
Ok,  no problem. From here it doesnt need any mention  its now standard lol

martinxyz, 2016-01-07:
The feature has a disadvantage, too - mechanical slack will be more notable, because the position error depends on the direction.

openhardwarecoza, 2016-01-08:
True, but theres just no excuse for building a machine with slack ;) tune acceleration, design properly )Vslot instead of nylon rolers on mitsumi lol flat edges) etc

openhardwarecoza, 2016-01-08:
Did some rather massive cleanups today

Removed all the unused copies of jquery that landed with each copied in section lol.

I also moved the paperscript out of index.html into a external script

openhardwarecoza, 2016-01-08:
I sat down with my test file, and looked at the precision of each move, each pixel, and quickly discovered between the left->right and right_>left strokes, I was off by 3 pixels in X. 

First off all I disabled the optimisation, to rule that out

Then I temporarilly added the canvas for paper.js, to rule that out (i thought it was my gcode generator code) - and immediately saw it was in there already. 

I do the draw in the iteration loop so pretty much changed a lot! 
I didnt go scientific on it (got a tooth extracted day before, feeling a bit feverish today so not in prime thinking condition) instead I trial-and errored - until both rows start, finish and all objects in them - all lined up perfectly

For example here https://github.com/openhardwarecoza/LaserWeb/blob/f44ebadc3537bd45458b4ff1173c1aa815e1058c/i/raster/laserraster.js#L100 - i run the for loop to raster.width + 1 else it misses the last pixel in the row 
and
https://github.com/openhardwarecoza/LaserWeb/blob/f44ebadc3537bd45458b4ff1173c1aa815e1058c/i/raster/laserraster.js#L178 I need to take out a pixel - i think it relates to the from-to of the gcode (its not a dot but a line, so i think i start off by one pixel to the right of where I should be, then move TO the first pixel to get gcode for the first pixel.  Still, you clever okes could probably improve the way it was done! 
As of now though, it works!

Test file:
[image]

Now the lines line up nicely!  Ends are square, and dimensionally its accurate
[image]

openhardwarecoza, 2016-01-08:
Oh, forgot to mention, but I disabled the optimasation - when I put it back, looks weird again. Need to think about that one. 
first pixel, save gray value
-- lastgray =grayvalue (write no gcode)
second pixel:
-- If grayvalue == lastgray, start counting (write no gcode)
-- if not same, write the previous pixel(s) to gcode
-- lastgray = graylevel
third pixel:
-- If grayvalue == lastgray, start counting (write no gcode)
-- if not same, write the previous pixel(s) to gcode
-- lastgray = graylevel

etc

Something like that. That first ignore pixel will always need to be added (left to right) or substracted (right to left) from the X position when writing the gcode (so like s+= 'g1 x'+(x + offset)

martinxyz, 2016-01-09:
@openhardwarecoza it's called rubber duck debugging, great methodology: http://lists.ethernal.org/oldarchives/cantlug-0211/msg00174.html

openhardwarecoza, 2016-01-09:
Fixed a bug with newline Y moves this morning

openhardwarecoza, 2016-01-09:
Ps @martinxyz rubber ducky ordered!  Lmfao

openhardwarecoza, 2016-01-09:
@martinxyz  or a Stuffed Banana   

http://i.imgur.com/v4w0u.png

openhardwarecoza, 2016-01-10:
Okay, optimisation is back, @nathanielstenzel did some code optimisation, etc.  At this point I consider it "implemented" - will open extra issues for bugs as they come along (:

closed 2016-01-10 by openhardwarecoza