LaserWeb/deprecated-LaserWeb1 #33: [Feature Request] Webcam / Red Dot Layout Rotation

iceblu3710, 2015-12-31:
My #1 complaint with my current laser is aligning things is a PITA. I currently put alignment marks on multi cut pieces and need to pin and pivot the piece to keep everything aligned. It would be super helpful if I could jog the machine to (point1) and select that location on the drawing then jog to (point2) and select that in the drawing and the software would rotate the whole drawing to match.

Another feature would be to have a webcam facing towards the workpiece on the X/Y carriage and capture a video scanning the entire printable area. Then stitching it together and showing it as the background. This could then be used for the feature above but would not require any feedback for position info.

openhardwarecoza, 2015-12-31:
I like the alignment idea.  Fiducial Marks as its called,  my vinyl cutter uses that to line up preprinted prints for cutting...  Keeping this open (:

martinxyz, 2015-12-31:
In our fablab we have two fixed red lasers projecting a line on the X and Y axes of the bed/workpiece. Alignment has become so easy that it's a non-issue - it feels like cheating.

iceblu3710, 2015-12-31:
The problem is when you need to align a piece in two axis. Currently i align the piece lower left to a mark and zero the lazer. Then in Corel Draw i need to take the (x, y) offset from that corner to the upper right marker. I then move the laser back/forth to get both locations under the alignment laser.

Takes 10x longer to align than cut.

openhardwarecoza, 2016-01-05:
@johnlauer is going something similar in Chilipeppr - where he uses the camera widget to view the fiducials, then wants to scale and rotate the code to match  the actual PCB angle/size.   We may be able to share some workload

openhardwarecoza, 2016-01-10:
I'll close this one for now (:

closed 2016-01-10 by openhardwarecoza