LaserWeb/deprecated-LaserWeb1 #34: DFX Error loading gears

cojarbi, 2016-01-01:
Not sure if it affects the process itself but i get this when prepare for laser is pressed

Min Dimensions
X: 0 Y: 0 Z: 0
Max Dimensions
X: 0 Y: 0 Z: 0
Total Dimensions
X: 0 Y: 0 Z: 0

Opening OpenJSCAD
Min Dimensions
X: 100000 Y: 100000 Z: 100000
Max Dimensions
X: -100000 Y: -100000 Z: -100000
Total Dimensions
X: -200000 Y: -200000 Z: -200000
Incoming file Converted to GCode...
Gear Generator Complete...
Parsing DXF:...

DXF Errors!:
Invalid Entity: eof

NB! There were errors while parsing the DXF. Usually this is normal, unsupported elements are not processed. Check render before cutting...
Min Dimensions
X: 0 Y: 0 Z: -3
Max Dimensions
X: 101.1161137261586 Y: 81.14400767290104 Z: 0
Total Dimensions
X: 101.1161137261586 Y: 81.14400767290104 Z: 3
Incoming file Converted to GCode...
DXF File Upload Complete...
Parsing DXF:...

DXF Errors!:
Invalid Entity: eof

NB! There were errors while parsing the DXF. Usually this is normal, unsupported elements are not processed. Check render before cutting...
Min Dimensions
X: 0 Y: 0 Z: -3
Max Dimensions
X: 101.1161137261586 Y: 81.14400767290104 Z: 0
Total Dimensions
X: 101.1161137261586 Y: 81.14400767290104 Z: 3
Incoming file Converted to GCode...
DXF File Upload Complete...

openhardwarecoza, 2016-01-01:
Cosmetic... OpenJSCAD adds the word 'eof' to signify End of File. 

Millcrum.js just ignores it.  Give me 5 mins and I'll add some error handling just to not show that (:

openhardwarecoza, 2016-01-01:
Fixed:  
https://github.com/openhardwarecoza/LaserWeb/commit/bc7163baafdd2ed95be4418c9ac132a34dd64c7f

closed 2016-01-01 by openhardwarecoza