LaserWeb/deprecated-LaserWeb1 #35: Vagrant 'ubuntu/trusty64' Procedure

iceblu3710, 2016-01-01:
I tried to install on my current system and node was seriously messed up so to eliminate my system errors I tried Vagrant. This is a clean testing environment:

````bash
$ vagrant init
$ vagrant box add ubuntu/trusty64
````  

Edit Vagrantfile:
````
config.vm.box = "hashicorp/precise32"
config.vm.network "public_network"
````

````bash
$ vagrant up
$ vagrant ssh
$ sudo apt-get install nodejs nodejs-legacy npm build-essential git
$ git clone https://github.com/openhardwarecoza/LaserWeb.git
$ cd LaserWeb
$ npm install
$ nodejs server.js
````  

---

### dotenv needs to be added to the npm install on the readme

Procedure courtery of @quillford : https://github.com/openhardwarecoza/LaserWeb/pull/10

Next install the npm modules by entering the following `npm install serialport socket.io node-static dotenv`

openhardwarecoza, 2016-01-01:
Could you be so kind as to do a pull request?  Or if you want the lazy way ojt,  while viewing readme.md in the browser,  there is a little edit pencil button next to the file name,  click,  add/fix procedure,  then submit (:

openhardwarecoza, 2016-01-01:
I am sadly on my phone,  away from pc   :(

closed 2016-01-01 by iceblu3710