LaserWeb/deprecated-LaserWeb1 #38: Scale, Rotate, Move, Bounding Box errors

funinthefalls, 2016-01-02:
The entry boxes for scale and rotate do not display any values typed into them. Bounding B0x button appears to do nothing. Move entry boxes only display one digit (45 will display as 5). Move on the X axis appears to move 10X the entered value i.e. enter 20 and it moves 200.

funinthefalls, 2016-01-02:
Update....increased my screen resolution to 1600X900 on a 32" Samsung monitor....now displays all entry boxes and buttons correctly. Only problem left is that the bounding box button appears to do nothing. Is there any way to get everything to display at a lower resolution? At 1600X900 text is very small.

openhardwarecoza, 2016-01-03:
Bounding box sends gcode to the machine (doesnt do anything onscreen)  

Are you using chrome,  i run it even on my tablet,  its not a resolution issue

openhardwarecoza, 2016-01-05:
Ok,  i managed to reproduce it on Chromium on my ubuntu Laptop

openhardwarecoza, 2016-01-07:
[image]

@funinthefalls - I did some resizing of the objects in that area, see if this resolved it for you?

funinthefalls, 2016-01-07:
Still the same....here is a screen shot.

Date: Thu, 7 Jan 2016 02:45:29 -0800

  
  

	
		
			
		
		This email has been sent from a virus-free computer protected by Avast. www.avast.com

openhardwarecoza, 2016-01-07:
I dont see the changes.  You did update?   I cant magically deploy it down to you (; 

Do a 'git pull'  in the laserweb directory

funinthefalls, 2016-01-07:
I did update it, I deleted the previous version completely, then dis a git clone.

Date: Thu, 7 Jan 2016 07:06:15 -0800

openhardwarecoza, 2016-01-07:
Ok.  Will see how else i can shuffle it

openhardwarecoza, 2016-01-10:
[image]

commit ce75727d (2016-01-10): Split middle toolbar into three rows to fix  #38
closed 2016-01-10 by openhardwarecoza