LaserWeb/deprecated-LaserWeb1 #39: Raspberry Pi - What version of NodeJS?

Protoneer, 2016-01-03:
Hi @openhardwarecoza ,

I am trying to build a RPI image with LaserWeb as part of the installed packages but I am having issues installing it. I currently have Node 0.12.6 installed...

dotenv does not seem to want to install on my version of node. Do you have a recommended version?

openhardwarecoza, 2016-01-03:
A long as the work i do doesnt  become a protoneer product like how you took andrews grblweb's ui,  and my grblpis from the opencncza project without attribution (;  (havent checked if attribution has been added since,  but your comments the last time I raised it with you wasnt very community like) -  i do welcome your support and working with laserweb otherwise (fork and branch,  but do give back too -  dont just take)  

Anyway,  if dotenv is giving issues on Pi,  its being used too critically yet,  so look at the commit from https://github.com/openhardwarecoza/LaserWeb/commit/ec80c71976782506bc16b084c35e88c3647685ed

I havent tested on a Pi since Dotenv was added

openhardwarecoza, 2016-01-03:
https://github.com/Protoneer/RPI-CNC-Config-Scripts/blob/master/scripts/wip-006-install-laserweb.sh   totally happy with what you are doing there (:

closed 2016-01-03 by openhardwarecoza

Protoneer, 2016-01-03:
Thanks @openhardwarecoza . Things will be done better this round. (Not really the place to ask, but have you played with CNC.js? I was pleasantly surprised how polished it is)

openhardwarecoza, 2016-01-04:
Excellent, then I look forward to working with you (:

I have looked at CNC.js indeed.  I don't totally like Andrew Hodels serialport code (connects to ALL ports on startup), so found cnc.js while I was looking for a better node.js/SerialPort implementation example to build from.  Been meaning to get to porting some of that over from cnc.js into Laserweb (being able to select the baud rate and a connect/disconnect button in the browser feels like it would eliminate some of the support Issues I get here. Put it on hold pending #21 - but if there isnt much progress on #21, I may p[roceed with that anyway. 

Other points:
I think my 3D viewer looks better (i hate that cnc.js, chilipeppr, all draw the negative quadrants too. Its lazy (thats the default three.js config) - yet almost all our cam's and machines operate only in the first quadrant (not all, I know, but 80% of the maker people.  I often 0,0 in the center of a workpiece - but still for most use cases, not) - especially in that the axes scales show you at least that its as large as you think it is. With the others you dont get that visual clue

What I do love about cnc.js is a) the polish, it really is nicely polished, but then again they are 750 commits ahead of me (: - he is grafting though! Doing this full time? In between the dayjob I can't turn out that many commits in 4 months... and b) the gcode widget with the line numbers and status - I am considering porting that too. 
The rest of the features are for spindles/cnc work - so not really of an interest to Laserweb at the moment, but I am definately always keeping cnc in mind in any change (even though lasers arent really Z aware I keep all Z code so we can use it with CNCs too, in the DXF via MC code, I still kept the tabs and z code too so one can use that in browser to make cnc gcode, I also kept the inside/outside cuts code (tool diamter offset) 

I can see a near future where Laserweb has a tickbox for "is CNC, is 3DP or is Laser" in its config