LaserWeb/deprecated-LaserWeb1 #41: commend get doubled

cojarbi, 2016-01-03:
When a send a command like G0 X300 F5000, it send it twice. you can see it in the log window

openhardwarecoza, 2016-01-04:
It definitely only gets sent once. If you see it twice in the console, it could be that your firmware has Command Echo turned on (its a debugging thing, turn it off, its a waste of bandwidth)

(i.e I print what I sent, and then the firmware echo's it back again, which shows as a second entry in the log. 

Can't give you a more exact instruction unless you elaborate on what you are running?

cojarbi, 2016-01-04:
Would this echo make it go twice the distance?  When I run this it goes 600.

openhardwarecoza, 2016-01-04:
I already asked above for details:

"Can't give you a more exact instruction unless you elaborate on what you are running?"

-  Firmware used?
- Screenshots of issue? 
- Exact step to reproduce (are you sending G0 X300 from a gcode job, or by manually typing into the >_console block?

cojarbi, 2016-01-05:
Not happening anymore, not sure why but can reproduce it

closed 2016-01-05 by cojarbi