LaserWeb/deprecated-LaserWeb1 #42: Arcs are being processed incorrectly/ignored in DXFs.

prcdslnc13, 2016-01-04:
Arcs and trimmed circles in polylines come out with incorrect radii or are skipped entirely when opening .dxfs. The .dxfs are generated in the latest version of DraftSight on windows 10. 

[arc-test.zip](https://github.com/openhardwarecoza/LaserWeb/files/77515/arc-test.zip)

openhardwarecoza, 2016-01-04:
Thanks,   will invesigate, would be in millcrums dxf parser,  dxf.js.

prcdslnc13, 2016-01-04:
[acad output.zip](https://github.com/openhardwarecoza/LaserWeb/files/77792/acad.output.zip)

Tried a drawing from autocad and got similar but actually less consistent results. 1 of the arcs at the lower right corner actually output correctly, which is interesting.

openhardwarecoza, 2016-01-05:
Ties in with https://github.com/openhardwarecoza/LaserWeb/issues/27 to a degree

openhardwarecoza, 2016-01-05:
My personal workflow is sketchup -> Guitarlist plugin as polyline dxf -> Laserweb
Sketchup renders arcs and circles as line segments anyway - always much easier.  Arcs is tough, since the standard is "each has their own standard"

openhardwarecoza, 2016-01-05:
https://github.com/openhardwarecoza/LaserWeb/blob/master/i/millcrum/inc/dxf.js#L201  would be where to start digging, along with http://www.autodesk.com/techpubs/autocad/acad2000/dxf/arc_dxf_06.htm and some sample files to check.  Anyone want to help?

openhardwarecoza, 2016-01-06:
@prcdslnc13, in autocad, can you try https://knowledge.autodesk.com/support/autocad/learn-explore/caas/CloudHelp/cloudhelp/2016/ENU/AutoCAD-Core/files/GUID-0C422AA9-23DD-4650-AD66-68E9D7989E3F-htm.html = split the Arcs into Polylines?  Revert with feedback

I fixed a bug in the 3D viewer's handling of G3 arcs, but still looks like a DXF parser issue in your case

openhardwarecoza, 2016-01-06:
@buserror - I note you too have been playing with Millcrum.

Any chance you can assist in fixing some bugs in the Arc handling - Andrew doesnt want to spend any time on it anymore and I don't have the skills

prcdslnc13, 2016-01-07:
@openhardwarecoza that is actually what I do for all these. It makes LaserCAD and most of the CNC software I use work right. Ill pull down the latest version, Im curious what happens when I explode the lines and what they look like. But right now everything is a poly line.

prcdslnc13, 2016-01-07:
@openhardwarecoza exploded lines and arcs dont process at all so thats a bad plan. I did a quick try of different dxf versions. r12 2000-2002, 02-04, and 2013 in both binary and ascii. Binary didnt process at all in any of the versions. R12 didnt bring the polylines over. The 00-02, 02-04 versions had the issue I sent earlier. I typically use 00-02. The 2013 failed. I can try again and copy some of the errors if its useful. Sorry I looked at the code but I am not a coder by heart so its a bit over my head. I sent it to a fiend of mine thats a big js guy. He may or may not be able to help.

cojarbi, 2016-01-07:
Some DXF with arcs, done in inkscape. Export via regular save as and openscad dxf output plugin
[Archive.zip](https://github.com/openhardwarecoza/LaserWeb/files/80751/Archive.zip)

buserror, 2016-01-07:
@openhardwarecoza well not sure, I submitted a small merge request a few weeks ago and he hasn't replied, or merged so I don't see the point at the minute..

openhardwarecoza, 2016-01-07:
Thanks @buserror , sadly https://github.com/openhardwarecoza/LaserWeb/issues/27#issuecomment-167587329 is the reason I guess. Well, I'll follow you here (: If you do anything cool with CAM let me know (:

openhardwarecoza, 2016-01-09:
Ok, so today I sat down and started playing with this:

1.  I added https://github.com/gdsestimating/dxf-parser
2.  I set it to run when a DXF is opened  https://github.com/openhardwarecoza/LaserWeb/blob/master/i/main.js#L1192 (before the cutting params / gcode step here https://github.com/openhardwarecoza/LaserWeb/blob/master/i/main.js#L1290 -> https://github.com/openhardwarecoza/LaserWeb/blob/master/i/main.js#L1298) just to see what it parses
3.  Initial impression, AWESOME: Picks up my most complicated DXF and draws it perfectly;

[image]

4.  I compared the raw data from both MC's dxf parser and from the new one:
https://github.com/openhardwarecoza/LaserWeb/blob/master/i/dxf/renderer.js#L175   and from https://github.com/openhardwarecoza/LaserWeb/blob/master/i/millcrum/dxf.js#L262

Both find the same data ... so millcrum's dxf.js isnt broken too badly in the parsing department - it is lacking a few entity types though so I think we will move away from it still if possible (plus upstream support is a big plus we need - someone who actually merges pulls, and helps fix bugs)

Where MC then presumably chokes completely would be in the code where it takes 'arcs' and converts them to 'polylines' or rather a bunch of G1 moves.   We don't really need that, I think.  Grbl, Marlin, Smoothie all have G2/G3 arc support?  I presume MC used the lines to more easily calculate paths (since the original purpose of dxf.js was to parse dxfs to millcrum for use on millcrum.com - not really as a direct to gcode generator)

So to push forward with this we need to take: https://github.com/openhardwarecoza/LaserWeb/blob/master/i/dxf/renderer.js  and  for each of:

https://github.com/openhardwarecoza/LaserWeb/blob/master/i/dxf/renderer.js#L113
https://github.com/openhardwarecoza/LaserWeb/blob/master/i/dxf/renderer.js#L173
https://github.com/openhardwarecoza/LaserWeb/blob/master/i/dxf/renderer.js#L259
(Text is a tricky one - easy to do in three.js but not easy to do in gcode)  https://github.com/openhardwarecoza/LaserWeb/blob/master/i/dxf/renderer.js#L241
(solid - I dont think we want to support that one)  https://github.com/openhardwarecoza/LaserWeb/blob/master/i/dxf/renderer.js#L204

we need to just add gcode writers (currently it creates three.js objects)

simply gcode += 'G.......F....' is all I need to integrate it deeper into laserweb...

My new setup would then be (since this can draw straight to three.js)

1. Open DXF 
2. Run the parser and draw these to three.js, each layer in a different color!  Also at this time generate the gcode moves and store with placeholders for laser power/speed (s+= 'G1 ....' )
3. Open the cutting params modal (its draggable) and you can see the colors on the layers as well as the canvas - so you can visually select params to match for the layers you want
4.  When you click Prepare for cutting, we parse the s= '    ' data and replace the placeholders for S???? and F????  with the values selected in the modal (just like we do now - except now we place those into MC code then run MC to convert that to gcode.  If we have gcode already in step 2 we save a step! and several seconds) 

5.  Later we can add optimisation like traveling salesman. For laser even keeping "paths" in an object is pointless - its not like a CNC that run in paths to do chip clearing etc - so technically we could just do nearest next G1 move (;   - realistically though we'd start from the inside and work our way out - small circles first - but thats laters worry

openhardwarecoza, 2016-01-09:
@bzuillsmith   - we love your library!

openhardwarecoza, 2016-01-09:
@cojarbi - this new library even opens the Inkscape DXFs!  WOW!

[image]

openhardwarecoza, 2016-01-09:
[dxfs.zip](https://github.com/openhardwarecoza/LaserWeb/files/83770/dxfs.zip)

heres a few trouble DXFs you can test with - laserweblogo works in Millcrums doesnt work in dxf-parser+https://github.com/gdsestimating/three-dxf)

The rest works not at all in millcrum (or works but displays totally wrong) but draws really nicely in dxf-parser+https://github.com/gdsestimating/three-dxf)

cojarbi, 2016-01-09:
Will try it today

openhardwarecoza, 2016-01-09:
@cojarbi read the rest of what i wrote above it.  Nowhere near testing phase.  We know it can read it.  But its not spitting gcode just yet

bzuillsmith, 2016-01-09:
@openhardwarecoza Thanks! Haven't had as much time to continue developing three-dxf (just had a baby!) but if you post some issues, I will certainly continue to do so when I have time. I also take pull requests. Posting some issues to dxf-parser and three-dxf would also motivate me to continue the work. I think it mainly handles lightweight (LW) dxf entities and not the more complex stuff yet.

openhardwarecoza, 2016-01-09:
Congrats on the little one!  Mines turning 6 this year.  Sleep does return!  But i will be very forgiving of slow Replies  (: 

Your lib is knocking the socks of the one we use at the moment.   Just need to sit down,  crunch through,  and add the gcode writers.  But having three.js in there (and its already a big part of laserweb) just fits the two together like a glove!

openhardwarecoza, 2016-01-09:
Ps:   if you have a to do list,  open an issue on your projects.  We have around 10-12 people here contributing code and about 20+ testing (:  -   would be beneficial to plough back some effort to you too (:

bzuillsmith, 2016-01-09:
Great, Just opened a couple. The biggest ones are about block insert support. Currently they are not supported, but I don't think it would take too much effort to get some basic block support going.

bzuillsmith, 2016-01-09:
PS: I also enjoy pair/mob programming, so if any of you would like to do that, you can find me on Google Hangouts

openhardwarecoza, 2016-01-09:
(: better than teaming up with a rubber ducky

openhardwarecoza, 2016-01-09:
https://github.com/gdsestimating/three-dxf/issues/1  explains why laserweblogo.dxf in the zipped samples above doesnt show up (:   -  thats just polylines

kaosat-dev, 2016-01-09:
Also started taking a closer look at the code & issue, thanks for the detailed writeup & examples !

openhardwarecoza, 2016-01-12:
I finally got the SVG issues sorted, so hopefully I can start looking at this one soon (:

openhardwarecoza, 2016-01-21:
Hi All.

Just checking in if anyone has made any headway with adding gcode output 

(https://github.com/openhardwarecoza/LaserWeb/issues/42#issuecomment-170259893 relates) 

@kaosat-dev @bzuillsmith 

We have a potential sponsor for an extra laser diode, @cojarbi approached JJohnson@jtechphotonics.com - hold thumbs. It could go to the community help polish out this feature

openhardwarecoza, 2016-01-21:
I've commented out the new lib for the time being - new users got tripped up when they see a render. 

As a side note, any Fusion 360 users who can test drive this and report back?  

https://apps.autodesk.com/FUSION/en/Detail/HelpDoc?id=appstore.exchange.autodesk.com%3Adxfsplinetopolyline_windows64%3Aen

athulsnair, 2016-01-22:
Got an error while cutting a Dxf file

Machine stopped during the 3rd pass
error: Bad number format"

dxf file generated using sketchup and scaled down to 20% in laserweb

@openhardwarecoza  that app didn't work for me, I will try again

openhardwarecoza, 2016-01-22:
Thanks for the feedback (:  - would have been nice if it did.  No worries, @bzuillsmith 's three-dxf has no problems reading it, so we just need to add the gcode writing (;

I'll check the circle shortly

openhardwarecoza, 2016-01-22:
Something is changing the S param during scaling.  Bug indeed but not in the DXF, thats in the transform tool!  Will fix asap

openhardwarecoza, 2016-01-29:
I have asked @johnlauer over in thread on https://groups.google.com/forum/#!topic/chilipeppr/VWho6tuvLJM about the possibility of working together on getting dxf support into CP (first) 
-  Why CP first? Well John already mentioned he may be able to three.js-> gcode (and up to here we have perfect dxf -> three.js that I can whip into a Chilipeppr widget in no time)    Putting the two together, would immediately benefit both projects!

openhardwarecoza, 2016-01-30:
@kaosat-dev @bzuillsmith  -  i had a sponsor approach me today and he offered a new 40w CO2 Tube + Mirrors to me to donate to a contributor.    As soon as he offered,  this issue sprung to mind.    Any chance either of you would be up for some free bounty in exchange for adding gcode output to three-dxf (or a new library based off it)  -  we have so much success parsing test dxfs with @bzuillsmits 's dxf-parser and three-dxf,  its really just gcode output thats missing.

funinthefalls, 2016-01-31:
If your bribe <grin> of the 40W tube is not wanted, I could really use a new tube, mine has just about completely failed. Can barely engrave, let alone cut anything. I wouldn't need the mirrors, my mechanicals are all working good.

openhardwarecoza, 2016-02-29:
Hi Guys

+jlauer released a new SVG -> Gcode widget for Chilipeppr today that contained some golden code! A function to take three-js geometry from a three-js group and convert it to gcode!   See https://github.com/chilipeppr/widget-svg2gcode/blob/master/widget.js#L452

This was part of the missin link we have had so far.  +bzuillsmith 's library already works perfectly to get dxf data into three-js geometry.  Three.js already contains code to do the heavy lifting of breaking arcs/circles into line segments (https://github.com/mrdoob/three.js/blob/ddab1fda4fd1e21babf65aa454fc0fe15bfabc33/src/extras/geometries/CircleBufferGeometry.js - also a big thanks to Ben for pointing this out to me) so with John's code, the magic storm is brewing and we are very very close to having the best DXF -> gcode parser the world has ever seen!

So far I have a tat for tat equivalent to the old millcrum code so good riddance to that (; - I'll make the frontend switch today 

- Equiv in that we can process polyline DXFs flawlessly with the new code I commited above...

What is still broken is that the line segments in arcs/circles above are drawn relative to point 0,0,0 at the moment - and LocaltoWorld() function of DXF doesnt seem to be able to find their world coordinates. 

Investigating, but i am already celebrating. Fixing this will be easier than fixing the old millcrum code, and most of the hard work of a new DXF library is done

funinthefalls, 2016-02-29:
Great job Peter! This was an important function to get working, and it seems that you are almost all the way there.

prcdslnc13, 2016-02-29:
This is awesome news! I've been head hunting people but have had issues
finding people that are as interested in this stuff as we are. Glad you are
making headway Peter!

openhardwarecoza, 2016-03-01:
And this now LIVE!

I temporarilly disabled OpenJSCAD (will switch it to the new toolchain over the next few days) - but please jump in and start testing DXFs! 

If you find funnies, post the DXF in a new issue.  Closing this Issue since it is now WRITTEN!

PS: You all owe +Johnlauer a beer!