LaserWeb/deprecated-LaserWeb1 #43: Raster processing times out.

dreydean, 2016-01-05:
Every attempt to include an image to raster and convert to gcode Chrome ends up timing out and I have to kill the process. Am I just not waiting long enough?

openhardwarecoza, 2016-01-05:
Try sticking to sane sizes,   grayscale,  600px.  Prep it in gimp if you have to

openhardwarecoza, 2016-01-05:
The new UI I just uploaded has a Slider for zoom.  Realistically with a diode laser you have around a 0.5mm spit, with a CO2 a 0.3mm spot (for engraving - some people even purposely un focus a little to get softer results)

So for example if you have a 400px wide image, that would come out as a 200mm wide engraving in real life with the spot set to 0.5mm.   The new slider shows you what it would be.

Use this to decide whether the image you are trying to use may be too high resolution.  If you need to start specifying a resolution smaller than the actual spot of the beam, it ont really work well in real life.   If thats the case, resize your image first then try again.  Remember, all this runs in the browser (in chrome's javascript engine) so a 3000x2000 pixel image may make chrome run out of memory.

[image]

openhardwarecoza, 2016-01-05:
https://html.spec.whatwg.org/multipage/workers.html may be something to investigate in future...

closed 2016-01-05 by openhardwarecoza