LaserWeb/deprecated-LaserWeb1 #44: Unable to move raster in Y axis

cojarbi, 2016-01-05:
I can move the preview in X but not in Y
See video https://plus.google.com/+ArielYahni/posts/Rz5ezxhRvYN

cojarbi, 2016-01-05:
I added my own image and it came in in negative Y, but i waas able to move it in positive Y. Note the case when imported the gcode you sent
[image]

openhardwarecoza, 2016-01-05:
Thats one of the known bugs I mentioned on Google Plus last night.

There was three, I tacklet the laser power scaling bug first this morning, havent gotten to that one yet.  The bug is that the Physical image size (needed to offset it from the negative) only gets calculated ff you change the Spot Size  before clicking Raster Now

openhardwarecoza, 2016-01-05:
And this bug is fixed:

https://github.com/openhardwarecoza/LaserWeb/commit/a77c3833b8f430e7794526fced8bd129548f6475

openhardwarecoza, 2016-01-05:
[image]

Simplified the UI while I was at it...

closed 2016-01-05 by openhardwarecoza