LaserWeb/deprecated-LaserWeb1 #47: Calibration Widget

openhardwarecoza, 2016-01-06:
Remember when you got an Inkjet printer in the 90s, and it had you always go to do a calibration:

[image]

and have you print something like this

[image]

Whereafter you tell the driver which looked best?

I am going to add the same thing to Laserweb:
A little test engrave and test cut job that you can run, pick the best,. enter it in the UI, and it will save your presets.   I will have you be able to run it against multiple materials and save each too?

Like?

cojarbi, 2016-01-07:
Like very mucho

openhardwarecoza, 2016-01-10:
I already drew the image to use in the calibration widget:

https://github.com/openhardwarecoza/LaserWeb/blob/master/i/raster/calibration.jpg

And we've been using it manually to tweak a few...

At the moment I was too busy putting out bugs, havent worked with any of this any further yet.  @nathanielstenzel  - we could use you here too:

Copy the Raster widget but instead of the onload bla bla, just have the image in place already, different title, and modal doesnt close (stays open, displays some advice, make it smaller so one can see 3D viewer better and copy Send to Laser button onto Div)

nathanielstenzel, 2016-01-12:
https://github.com/nathanielstenzel/LaserWeb/commit/06d21422f35a1700170602fa42cc3a4e20b237c9
That is a commit that has my start of the calibration widget so far. I tried to merge in your changes. I can't seem to make everything in the widget turn on.

openhardwarecoza, 2016-01-12:
That commit has some of my commits too :|    i need to show you how to handle remote git branching so you can seperate the pull of my work,  from the commit of your work (:

All the elements on the widget (html)  has accompanied objects in main.js (jquery)

nathanielstenzel, 2016-01-12:
I may have actually given you the wrong link. Perhaps a slightly earlier commit of mine would have not had your changes in it. https://github.com/nathanielstenzel/LaserWeb/commit/2e3af070eaa5a37d73a7b2588b7527f81cdafe92

nathanielstenzel, 2016-01-12:
Yep. There you go. That one is all me. I need to get your new layout in there though.

nathanielstenzel, 2016-01-13:
https://github.com/nathanielstenzel/LaserWeb/commit/a983e0ea6ded08e923e0685ccfd48fe6aca5c324
That does not have your change in design of the button layout for the upper left, but it works. The calibration is done by the main raster widget. The code was not exactly reusable, so I just expanded it with hidden stuff. It has the side effect of the raster widget being centered. Feel free to change that. Next up would be to add other slide bars for speed vs darkness calibration. That is another chapter though.

nathanielstenzel, 2016-01-13:
[image]

nathanielstenzel, 2016-01-13:
And yes, I know you want the title changed. I just don't know how I want to go about doing that. I decided it was easier for me to expand the raster widget than make a new widget.

nathanielstenzel, 2016-01-13:
https://github.com/nathanielstenzel/LaserWeb/commit/385d98b22acff5186ce60c6fa791e519c90e7efc The Raster widget title now differs for the image mode and the calibration mode.

openhardwarecoza, 2016-01-13:
Ok, let me try and cherry pick from your commits (: 

Else I'll do copy paste and attribute lol

openhardwarecoza, 2016-01-13:
I had to go manual :) not even cherry pick worked
Luckilly it wasnt a lot of code to find.

I did however not use the dropdown - file button in a bootstrap dropdown looks a litle weird sadly :(

nathanielstenzel, 2016-01-13:
Got a screen shot?  You have me curious about how it looks and I will not be home for 5 hours.

openhardwarecoza, 2016-01-13:
[image]

nathanielstenzel, 2016-01-13:
I meant the button that gets you to the raster calibration mode.

openhardwarecoza, 2016-01-13:
Top left

nathanielstenzel, 2016-01-13:
Okay.  I see it.  Will that calibrate all laser output then?

nathanielstenzel, 2016-01-14:
The send to laser button should be locked when you open the window, but it seems I don't relock it when the window opens or you click the raster button. Two bugs for me to squash in my modification.

I will try to work on that in about 19 hours.

nathanielstenzel, 2016-01-15:
https://github.com/openhardwarecoza/LaserWeb/pull/56 That should squish those 2 bugs.

nathanielstenzel, 2016-01-15:
Please only use https://github.com/nathanielstenzel/LaserWeb/commit/177d2415f0570b2df6a7bf223a23f9801ca8ed0f for the pull for that bug fix.

nathanielstenzel, 2016-01-16:
You mentioned an interest in variable speeds for darknees control. How does this look?
[image]

openhardwarecoza, 2016-01-16:
Looks good!

nathanielstenzel, 2016-01-16:
This is now ready for testing.

closed 2016-01-21 by openhardwarecoza