LaserWeb/deprecated-LaserWeb1 #53: The upper left is cramped and the file icon does nothing?

nathanielstenzel, 2016-01-12:
After the SVG button was added, the RASTER and the GCODE buttons got cramped and not all of the letters are visible.

openhardwarecoza, 2016-01-12:
(: Buy a 1080p monitor (;

Just kidding - I'll have to make a plan there... The file icon is just  visual indicator that the ones to right means Open ... gcode, Open....svg, etc

openhardwarecoza, 2016-01-12:
[image]

Fixed 

(the STL(Soon)button is a mockup - coming soon something, may be stl may be something else, but same number of buttons in both rows just look so much better.  Hey we can put Calibration there?

closed 2016-01-12 by openhardwarecoza

nathanielstenzel, 2016-01-12:
I was going to have the calibration button be a drop down of the Raster like there is a drop down of the JSCAD. In fact, I already have the change committed to my branch, but I have to pull in your new layout and merge the changes into my branch.