LaserWeb/deprecated-LaserWeb1 #54: Smoothie with dual pin (non pwm)  supplies

openhardwarecoza, 2016-01-13:
Need to add m3/m5 in smoothie code too for users with two pin config (pwm into IN via capacitor + TH/TL on a smootie switch pin) 

Thanks to https://github.com/TurnkeyTyranny/laser-gcode-exporter-inkscape-plugin/pull/8 for bringing to my attention

ajfoul, 2016-01-13:
Please note that Smoothie config will be need to have a switch to enable M3/M5 to fire. I've got a [sample here](https://gist.github.com/ajfoul/b32a0432f337e81f3ab6) (pin assignments may need to be adjusted depending on board and wiring)

And, all the two pin PSUs I've seen do PWM on IN.  In some cases a logic level shifter is needed because some PSUs don't support Smoothie's 3.3v logic.

openhardwarecoza, 2016-01-13:
(: I have the level converters in my board, and we have it documented nicely on the SmoothK40 Wiki entry too:   http://smoothieware.org/blue-box-guide 

Thanks for the sample config. Will update the Wiki tonight

prcdslnc13, 2016-01-13:
This just answered my biggest question on how to wire all this up. Thanks
for the post.

openhardwarecoza, 2016-05-11:
Available as configurable option in LaserWeb2 - so closing this one

closed 2016-05-11 by openhardwarecoza