LaserWeb/deprecated-LaserWeb1 #55: The scroll bar on the right follows popup windows.

nathanielstenzel, 2016-01-14:
It does not seem to matter what type of file you open, if you get a pop up window, the scroll bar on the right will follow the pop up window. That means that it might hide out of sight until you move your pop up window to the left some. The scroll bar will keep a consistent distance from the pop up window.

openhardwarecoza, 2016-01-14:
Can you show me a screenshot annotated with where you mean exactly?  I tried with a small screen resolution, then on 1080p and neither here shows up as a clear issue?

nathanielstenzel, 2016-01-14:
[image]

nathanielstenzel, 2016-01-14:
That bar should be on the right, but since I dragged the pop up window, it is on top of the grid.

openhardwarecoza, 2016-01-15:
Ok, i can replicate. will see why that is

openhardwarecoza, 2016-01-15:
Tried a few things, but doesnt seem to be our side.

See if you can get it isolated into a JSFiddle then open a bug here https://bugs.jqueryui.com/newticket

nathanielstenzel, 2016-01-16:
Fixed in https://github.com/nathanielstenzel/LaserWeb/commit/d1dda5b6d35bcc9745110454c15bf936f9e3ffa0 courtesy of some code written by wellwind in https://gist.github.com/wellwind/a5a10222ad691d72a11c

closed 2016-01-16 by openhardwarecoza