LaserWeb/deprecated-LaserWeb1 #57: Frame around object on SVG load

cojarbi, 2016-01-15:
As i load and SVG a frame is made around the drawing. it seems as if laserweb is reading the page itself from inkscape. 
[test copy.svg.txt](https://github.com/openhardwarecoza/LaserWeb/files/91351/test.copy.svg.txt)
[image]

openhardwarecoza, 2016-01-15:
Yes, thats documented (which is why I added the DPI buttons to get a more accurate size

https://plus.google.com/+PetervanderWalt/posts/gXtqm5QRsBB  -  for under 600px wide SVGs - anything over I still have a bug I know where it is just not how to elegantly solve yet)

SVG sucks in that all positions are relative to the document.  Same is true for DXF.  Its great since it allows you to do multi pass jobs with a common 0 point.

1.  In your SVGs make your document properties the size of your laser (or the default 600x400 laserweb uses)
2. Position your artwork relative to bottom left corner 
3.  When importing to laserweb, dont bother with the slider - just use the DPI buttons to get it accurate

cojarbi, 2016-01-15:
Noted

openhardwarecoza, 2016-01-15:
I am going to add the dpi buttons to raster too.    Accurate dimensions will be easier than the sliders

closed 2016-01-15 by openhardwarecoza