LaserWeb/deprecated-LaserWeb1 #58: Updated install instructions for Windows users

funinthefalls, 2016-01-15:


funinthefalls, 2016-01-15:
Can you add this to the Readme file?

openhardwarecoza, 2016-01-15:
You can either post it as a pull request (in the web github.com,  open readme,  click the edit button. (little pencil)  and apply changes.  Submit as a pull request.  

Or i can give you contribution rights to the repo if you are familiar with git? 

Just a note,  dont download the zip,  make sure you install git instead.  The auto update components of laserweb uses git...  So skipping that,  breaks future work (:

funinthefalls, 2016-01-15:
The reason I said to use the zip instead of git was because these instructions were posted as a result of a user on the opensource laser group that asked for step by step instructions because he was not that computer savvy. So I figured there was no way someone who needs to have there hand held to install it would be able to install and use Git For Windows. I would not mind having contributor rights for the repository, I have a basic understanging of the Git workflow process, I just don't want to screw anything up by accident.

Date: Fri, 15 Jan 2016 11:56:09 -0800

  
  

	
		
			
		
		This email has been sent from a virus-free computer protected by Avast. www.avast.com

openhardwarecoza, 2016-01-16:
Collaborator status added (:   welcome to the team.  (: 
I love how LaserWeb is becoming community property

closed 2016-01-16 by openhardwarecoza

funinthefalls, 2016-01-16:
Thanks for the add Peter. Once LaserW0eb matures (shouldn't be too much longer with the hard work you have been putting in) we should tackle modding Repetier to a laser specific version.

Date: Fri, 15 Jan 2016 20:14:03 -0800

openhardwarecoza, 2016-01-16:
(: i was just 5 mins ago loading up Repetier to one of my Evitabot controllers so I can take the plasma for a spin on Monday when the neighbours go to work (wont piss them off by playing on a sunday).  Setting num_extruders to 0 disabled all the temp controls.  One line,  so clean and elegant.  Love that firmware @Repetier is a legend!