LaserWeb/deprecated-LaserWeb1 #62: Config.js file changes

funinthefalls, 2016-01-17:
I edited the config.js file in my local copy to reflect the bed size of the K40 (I can get 320X220mm on my machine. When I restart LaserWeb it still displays the 600X400 size for the render window in the center of the page.

openhardwarecoza, 2016-01-17:
Also change bed size in ui.js,  as mentioned in the readme

openhardwarecoza, 2016-01-17:
https://github.com/openhardwarecoza/LaserWeb/blob/master/README.md#config

Although,  it would be nice if we make it a global variable...

openhardwarecoza, 2016-01-17:
Also,  you'll notice that your next git pull update fails because of the config changes (since config.js is in the git repo too). 

https://github.com/openhardwarecoza/LaserWeb/issues/7 comes to mind to fix this problem.   In fact,  i think this one should be worked on urgently,  the more people start using laserweb,  the more per-user configs will need to be managed

openhardwarecoza, 2016-01-21:
I'm going to close this issue - since the relevant issue is #7

closed 2016-01-21 by openhardwarecoza