LaserWeb/deprecated-LaserWeb1 #63: Some buttons's clickable area is limited

chils41, 2016-01-17:
The following buttons clickable area is limited. Only bottom 30% portion is clickable. Its not clickable even on the text on that button. Not big issue but not everybody tries to click on every corner of the button to see if its working or not. 

Affected buttons are:

GCODE , DXF, SVG, REST 3D VIEW, MIRROR, FIX00, DRAW BOUNDING BOX, 

Looks line all those buttons are in same line are affected by something!

nathanielstenzel, 2016-01-18:
This might be related.
http://stackoverflow.com/questions/15859638/jquery-mobile-button-area-not-fully-clickable-only-left-part-of-button-trigger

openhardwarecoza, 2016-01-18:
https://github.com/thomaspark/bootswatch/

openhardwarecoza, 2016-01-18:
I know why!   I am shrinking the default paper:bootswatch header,  but there must still be hidden css overlaying on the buttons.  Will inspect and fix in a second

nathanielstenzel, 2016-01-18:
Weird, my link that I posted on the matter disappeared....

nathanielstenzel, 2016-01-18:
http://stackoverflow.com/questions/15859638/jquery-mobile-button-area-not-fully-clickable-only-left-part-of-button-trigger

openhardwarecoza, 2016-01-18:
Thanks for the superb error report @chils41 - your clue that triggered me into realising what it was, was your observation that they are all in the same 'row' so to speak.  That made me realise the Navbar may have something laying over that height:
Did an inspect of the object using chrome dev-tools and my hunch was confirmed:  Navbar is 64 px tall, depite me using navbar-xs @28px tall...

[image]

Fixed in 

https://github.com/openhardwarecoza/LaserWeb/commit/49f7a788c321949bada258e387cd884e6175e7ef

closed 2016-01-18 by openhardwarecoza

chils41, 2016-01-18:
Awesome! I dont know why I did not think of "inspect element". But its solved and I am happy!

openhardwarecoza, 2016-01-18:
Keep up your good work Without testing we can't move forward (: