LaserWeb/deprecated-LaserWeb1 #67: unable to install

athulsnair, 2016-01-19:
Hi,
I installed the software, but i coudn't genrate gcodes for .dxf files. I am attaching the dxf file generated gcode.
 I tried to install in anotherpc but getting the same resul.
I'm using Windows 7 64bit and GRBL
Also I can't control the power,can't scale or move.
Am I the only one having this problem

[image]
[Material Thickness.docx](https://github.com/openhardwarecoza/LaserWeb/files/95519/Material.Thickness.docx)
[image]

openhardwarecoza, 2016-01-21:
This also relates to https://github.com/openhardwarecoza/LaserWeb/issues/42

Please share a link to https://github.com/openhardwarecoza/LaserWeb/issues/42 on your social media and hunt down anyone you know that knows Javascript - we can always use more people to get the development out the door quicker (:

Basically - the current library works only with Polyline DXFs and doesnt support Arcs (so for example dxfs from openscad and sketchup work perfectly) 

The new library we have started integrating can parse and display pretty much any DXF I throw at it (displays it in blue) but doesnt yet generate Gcode.  For each of the three.js draw functions from https://github.com/openhardwarecoza/LaserWeb/blob/master/i/dxf/three-dxf.js#L287 downward - we just need to generate Gcode...  If you read through the original thread, you'd see there are a few people working on it (:

In the meantime, what do you draw in?   Does it have the option to break arcs/circles into polyline segments? And then of course export your files as polylines (not lines)

openhardwarecoza, 2016-01-21:
Closing this, rather follow along (and interact please) on issue #42

closed 2016-01-21 by openhardwarecoza

athulsnair, 2016-01-21:
The file taht you can see in the screenshot is generated using fusion 360.

openhardwarecoza, 2016-01-21:
Doesnt matter, same issue.  Set fusion to export polyline dxfs without arcs
or use the builtin cam

athulsnair, 2016-01-21:
Also the scaling and moving tabs are not working.

openhardwarecoza, 2016-01-21:
Listen to what I asked, go read the other thread.  The Blue one is just a debug.  Use a valid polyline DXF and you will see the difference...

openhardwarecoza, 2016-01-21:
Your log will also show the errors.

openhardwarecoza, 2016-01-21:
I disabled the blue debug render.  Now you'll see your dxf wont open ANYTHING since its not polyline

openhardwarecoza, 2016-01-21:
[polyline-dxf-from-sk.dxf.zip](https://github.com/openhardwarecoza/LaserWeb/files/98824/polyline-dxf-from-sk.dxf.zip)

Heres a plain polyline dxf with no arcs.    Give it a go (:

If you'd like us to support arcs and non-polyline shapes - head over to #42 and help us out (:

openhardwarecoza, 2016-01-21:
This should also help:  

https://forums.autodesk.com/t5/design-validate-document/export-svg/td-p/5812107

openhardwarecoza, 2016-01-21:
I am not a fusion user , but you could give this a try https://apps.autodesk.com/FUSION/en/Detail/HelpDoc?id=appstore.exchange.autodesk.com%3Adxfsplinetopolyline_windows64%3Aen

openhardwarecoza, 2016-01-21:
Test example with polyline dxf:

[image]

And scaled to 50%

[image]

cojarbi, 2016-01-21:
this is what i got with that 
[image]
[image]

openhardwarecoza, 2016-01-21:
That looks pretty usable?

openhardwarecoza, 2016-01-21:
but wait, thats blue not red. Do a git pull first (:

cojarbi, 2016-01-21:
updated my comment. above