LaserWeb/deprecated-LaserWeb1 #70: Send code

funinthefalls, 2016-01-19:
When using the send code option it double sends whatever is entered ie:
G90 sends:
G90
G90

openhardwarecoza, 2016-01-21:
Hmmm, @cojarbi once experienced this on his AZSMZ smoothieboard:
https://github.com/openhardwarecoza/LaserWeb/issues/41

But couldnt reproduce to test.

Are we sure it gets sent twice, and not a case of the board echoing back what it received?

cojarbi, 2016-01-21:
I think it's just echo, I wasn't able to reproduce the movement

Ariel Yahni

Mad-Onion, 2016-01-22:
I can confirm this issue.
I sent the $?-command just one time - here ist the corresponding log from the web-console with the command transmitted two times:
```
SEND: $?
$$ (view Grbl settings)
$# (view # parameters)
$G (view parser state)
$I (view build info)
$N (view startup blocks)
$x=value (save Grbl setting)
$Nx=line (save startup block)
$C (check gcode mode)
$X (kill alarm lock)
$H (run homing cycle)
~ (cycle start)
! (feed hold)
? (current status)
ctrl-x (reset Grbl)
SEND: $?
$$ (view Grbl settings)
$# (view # parameters)
$G (view parser state)
$I (view build info)
$N (view startup blocks)
$x=value (save Grbl setting)
$Nx=line (save startup block)
$C (check gcode mode)
$X (kill alarm lock)
$H (run homing cycle)
~ (cycle start)
! (feed hold)
? (current status)
ctrl-x (reset Grbl)
```

And this is the output from the node-console (also doubled):
```
WARN: Ignored:  Port /dev/ttyACM0 said:  $$ (view Grbl settings)
WARN: Ignored:  Port /dev/ttyACM0 said:  $&#035; (view &#035; parameters)
WARN: Ignored:  Port /dev/ttyACM0 said:  $G (view parser state)
WARN: Ignored:  Port /dev/ttyACM0 said:  $I (view build info)
WARN: Ignored:  Port /dev/ttyACM0 said:  $N (view startup blocks)
WARN: Ignored:  Port /dev/ttyACM0 said:  $x=value (save Grbl setting)
WARN: Ignored:  Port /dev/ttyACM0 said:  $Nx=line (save startup block)
WARN: Ignored:  Port /dev/ttyACM0 said:  $C (check gcode mode)
WARN: Ignored:  Port /dev/ttyACM0 said:  $X (kill alarm lock)
WARN: Ignored:  Port /dev/ttyACM0 said:  $H (run homing cycle)
WARN: Ignored:  Port /dev/ttyACM0 said:  ~ (cycle start)
WARN: Ignored:  Port /dev/ttyACM0 said:  ! (feed hold)
WARN: Ignored:  Port /dev/ttyACM0 said:  ? (current status)
WARN: Ignored:  Port /dev/ttyACM0 said:  ctrl-x (reset Grbl)
WARN: Ignored:  Port /dev/ttyACM0 said:  $$ (view Grbl settings)
WARN: Ignored:  Port /dev/ttyACM0 said:  $&#035; (view &#035; parameters)
WARN: Ignored:  Port /dev/ttyACM0 said:  $G (view parser state)
WARN: Ignored:  Port /dev/ttyACM0 said:  $I (view build info)
WARN: Ignored:  Port /dev/ttyACM0 said:  $N (view startup blocks)
WARN: Ignored:  Port /dev/ttyACM0 said:  $x=value (save Grbl setting)
WARN: Ignored:  Port /dev/ttyACM0 said:  $Nx=line (save startup block)
WARN: Ignored:  Port /dev/ttyACM0 said:  $C (check gcode mode)
WARN: Ignored:  Port /dev/ttyACM0 said:  $X (kill alarm lock)
WARN: Ignored:  Port /dev/ttyACM0 said:  $H (run homing cycle)
WARN: Ignored:  Port /dev/ttyACM0 said:  ~ (cycle start)
WARN: Ignored:  Port /dev/ttyACM0 said:  ! (feed hold)
WARN: Ignored:  Port /dev/ttyACM0 said:  ? (current status)
WARN: Ignored:  Port /dev/ttyACM0 said:  ctrl-x (reset Grbl)
```

openhardwarecoza, 2016-01-22:
Thanks for the detailed feedback - tagging as Bug then

openhardwarecoza, 2016-01-29:
Fixed!

closed 2016-01-29 by openhardwarecoza