LaserWeb/deprecated-LaserWeb1 #71: Chrome page unresponsive

cojarbi, 2016-01-19:
Im getting this almost 5 - 7 times before the final render will appear. The process is the standard one. Choose raster, change setting and press "prepare raster"
[image]

Im on OSX 10.11.3 with chrome Version 47.0.2526.111 (64-bit)

openhardwarecoza, 2016-01-21:
Well, we are kind of pushing the limits of in-browser data processing.

That raster code is a for-loop iterating over X * Y pixels (for example if the image is small like 640x480, thats 307,200 iterations.

For a large image like 4000x3000px it becomes 120,00,000 iterations.  

In each iteration it needs to calculate the laser power, whether to skip (part of longer row) or qrite (end of same grayscale row) as well as the gcode coordinates.  It also has to save each move to the gcode buffer...

Long story short, as long as you stick to the limits of what your PC can handle (I'm running on an i7 with 16Gb RAM, 128GGb SSD, using Chromium on Ubuntu - and have only had the chrome 'wait' on images over 1000px wide) it works reasonably.

Now, there is the option of offloading some of the processing back to server.js:
But look at the install procedure to get Paper.JS to work with Node.JS (Dependecy on Cairo Graphics:

https://github.com/paperjs/paper.js?files=1#installing-paperjs-for-nodejs-through-npm

Thats what I wanted to avoid - therefore opted to do it in-browser

openhardwarecoza, 2016-01-21:
Adding the enhancement flag - I am sure a little code optimisation could improve that

Cinezaster, 2016-01-21:
Could it be an option to do it line by line?
Also have memory problems on raster images.
Feedback on the process would also be a neat feature. 

Really like to help out, need more hours in a day.

openhardwarecoza, 2016-01-21:
That might help! Good idea... Do a line, callback ui, do another line...@nathanielstenzel

mostley, 2016-01-21:
A seperate service for rasterizing and gcode slicing would be useful (like mentioned above). 
Perhaps doing it with phantomjs would be an option to avoid nodejs vs paperjs issues?

mostley, 2016-01-22:
anybody tried this yet?

openhardwarecoza, 2016-01-22:
I have yes, its merged into master and working nicely (:

The top and footer of the Rastering modal is missing though (:

mostley, 2016-01-22:
really? how is that? So the title with: "Raster engraving" and the close button is missing?
I can see it though.
But I just saw that if you open the dialog a second time the progressbar is still visible. PR coming up.

openhardwarecoza, 2016-01-22:
(: No problem - i dont mind UI issues - those are always easy to fix.  

I get this on the first run too:

[image]

mostley, 2016-01-22:
you get this while rasterizing or without starting the process?

openhardwarecoza, 2016-01-22:
1. Refresh page:
First run:
Upon loading the image I get the familar raster parameters modal 
Once i click Prepare Raster, I get that (goes away once the gcode loads  - so working fine, looks  a little funny with the missing header and footer

If I then click raster, open file, I immediately get the Rasterizing+progress bar instead of the anticipated paramaters

mostley, 2016-01-22:
that should have been fixed with this one: https://github.com/openhardwarecoza/LaserWeb/pull/80

openhardwarecoza, 2016-01-22:
Had to add https://github.com/openhardwarecoza/LaserWeb/commit/3087da2c9764c9e31ce98f1cc3bc4dd1f4257f97 to stop the rastershroud on a second time opening the file (:

I found why the footer is hidden. Your <div id="rasterProgressShroud"> is below the footer, I'll fit it now (:

openhardwarecoza, 2016-01-22:
[image]
Fixed in https://github.com/openhardwarecoza/LaserWeb/commit/221e18b7284f74903957382dd1d7122ed602b19b and https://github.com/openhardwarecoza/LaserWeb/commit/888d1416f7fa696a9787eb5caadb246064009903

mostley, 2016-01-22:
great!

cojarbi, 2016-01-22:
Guys nice job. Im getting cero "wait" on raster. Mind that the raster time in console is completely off. Also when the raster finishes and i press calibrate i get the rasterizing windows still. Dont know if that was taken care of

openhardwarecoza, 2016-01-22:
Taken care of, do a pull (:

That time estimate is based on feedrate in the file, if your machine has feedrate limits thats lower than what we are feeding it could run out by a lot

cojarbi, 2016-01-22:
Machine you mean computer?  Don't think that's the issue. I7 16GB Nvidia 2GB. SSD

The estimate for rasterizing the lady said 2958ms and it took more than 5 minutes. 

Ariel Yahni

openhardwarecoza, 2016-01-22:
[image]

I think you are looking in the wrong spot:

Estimated Job Time (in black) (1:19:59 in my example above) is the time the job will run (large example here, at over 1h19min)
The 3D Render completed in 584ms is how long your browser took to convert the gcode into a displayethree.js object  (just as a debug / interesting fact - nothing to do with the job)

The fact that you sau it said  2958ms (ms being the key - i dont have any milliseconds in the job time calculation) is why I think you are looking at the wrong field (;

openhardwarecoza, 2016-01-22:
Interesting stat though:

If your raster convert of the lady took 2958ms (I think we can safely assume that was the Raster Completed value you mentioned) - and mine took 2053.  My machine is only a Pentium F3430 @3.3ghz, with 12Gb of RAM, and a measly Gefore 210 1GB gfx.   (: And I am about 50% faster.   Difference?  Running Chromium on Ubuntu (me) vs (you) on full chrome on I assume windows?

cojarbi, 2016-01-22:
I'm on OSX.  And I'm looking at the 3D render completed  in xxx ms. But the actual time I saw the render in the screen was more than 5 min

openhardwarecoza, 2016-01-22:
It can never be 100% accurate - because of machine acceleration - feedrate of 1000 doesnt run as 1000mm/min from the moment it starts moving to the moment it stops.  We assume that acceleration adds about 30% extra time and add that onto what we calculate, and gets pretty close (;

cojarbi, 2016-01-22:
That would be on the actual hardware lazerizing and I agree,  to many variables so the estimate is ok. I'm referring to the time it took to show the render on the screen.

openhardwarecoza, 2016-01-22:
thats reporting on the step it took to generate the Three.js object - from there parsing the buffer and other syncronous tasks go on in the background that we dont time that also adds time

openhardwarecoza, 2016-01-22:
Since we dont get anymore Chrome hangs, and this is going totally off topic, can I close this one?

closed 2016-01-22 by openhardwarecoza