LaserWeb/deprecated-LaserWeb1 #72: Weird raster render result

cojarbi, 2016-01-19:
Using standard procedure im getting the following result in the raster render and calibration. This is under the latest commit. Not connected to any board
[image]
[image]

nathanielstenzel, 2016-01-19:
This is my fault for not testing my code and referencing a variable that is not even initialized. I made an adjustment. Please test.

nathanielstenzel, 2016-01-20:
I confirmed earlier that it is fixed. I still do not have the results for my patch though.

closed 2016-01-20 by cojarbi