LaserWeb/deprecated-LaserWeb1 #73: M3 / M5 support for GRBL

Mad-Onion, 2016-01-21:
I'm currently using grbl 0.9j, which gets detected correctly in the gui.
When I try to laser a path extracted from a SVG-File, LaserWeb does not use M3 / M5 to enable or disable the laser but moves the Z-axis instead. This way my configuration won't work.

Is there a way to change this behavior? 

Best regards
Tobias

Cinezaster, 2016-01-21:
I have the same configuration.
I'm using M3/M5 to control the laser and S### to control the spindle pwm output for laser power.
There are two types of controlling the lasers.
* via Z-axis: like you would engrave with a v-bit (lots of bigger lasers use this approach) 
* spindle speed (laser diodes engravers like mr beam, Chinese $100 laser diode engravers) 

Both should be an options when generating g-code.

Standard gcode cnc controllers stop the movement on a M3 or M5 command to compensate for the spindle getting up to the required speed. One would add a dwell `G4 P##` to pause and wait till the spindle is up to speed. 
For this reason they used the height of the z-axis with a very high acceleration setting to combine these three settings (x,y,laserpower).
Using the spindle on/off and speed commands for controlling lasers would need a more custom gcode controller. These would allow you to set spindle pwm while keeping engraver in motion.

openhardwarecoza, 2016-01-21:
Thanks @Cinezaster  - that may explain the dots we get at the end of raster lines on Marlin.  So far we are running smoothly on Smoothie, LasaurGrbl - without that. 

More importantly - I dont want to accomodate every possible configuration with If/Else anymore (:

So I am moving things like this into Machine Profiles:  See https://github.com/openhardwarecoza/LaserWeb/issues/7#issuecomment-173659979

I'm closing this one for now - lets get #7 in place first

openhardwarecoza, 2016-01-21:
@Mad-Onion  - see commit above - added M3/M5 for svg2gode

closed 2016-01-21 by openhardwarecoza