LaserWeb/deprecated-LaserWeb1 #74: Machine-Control for GRBL0.9j incomplete

Mad-Onion, 2016-01-21:
When I open the Machine-Control-panel it looks quite different compared to the screenshots.
Missing:
- Zero Machine
- Motors Off
- Fan Control

Instead I have a buttongroup "Air Assist".

Best regards
Tobias

openhardwarecoza, 2016-01-21:
Yip, @cojarbi reported issues with Grbl 0.9J as well on the S val;ues (grbl G used S0-255, and grblJ uses 0-10000 by default)

Just playing catchup at the office, but will fire up a grbl 0.9j board and play with it, unless one of you can take a stab at it (: - always happy to merge a pull request 

PS if you get the air assist buttons, looks like main.js thinks your running lasaurgrbl  - send me paste of your node terminal log ?

cojarbi, 2016-01-21:
I also got the same interface with air assist when connected. If I remember correctly also got it with G version

Ariel Yahni

Mad-Onion, 2016-01-21:
@openhardwarecoza here is the terminal log:
`WARN: Ignored:  Port /dev/ttyACM0 said:
INFO:  Found device:  Arduino__www.arduino.cc_ /dev/ttyACM0 Firmware Detected: Grbl 0.9j
WARN: Ignored:  Port /dev/ttyACM0 said:  Grbl 0.9j [&#039;$&#039; for help]
WARN: Ignored:  Port /dev/ttyACM0 said:  [&#039;$H&#039;|&#039;$X&#039; to unlock]`

Because of the "problem" regarding the M3 / M5 spindle / laser issue I will go back to an older, modded grbl-version.

openhardwarecoza, 2016-01-21:
I'm going to close this issue, since it relates to the work in https://github.com/openhardwarecoza/LaserWeb/issues/7

I decided to accomodate more user preference, we should add machine specific (what is the command for laser on, what is the command for laser off) into the machine profiles.   I'll continue work on that over the next few days - probably till over the weekend.  I pushed the initial work over in the meantime

openhardwarecoza, 2016-01-21:
Air Assist is actually correct:  See https://github.com/openhardwarecoza/LaserWeb/blob/master/i/main.js#L1956  (if grbl we show the air group) - I just never got round to adding the fuctions. I wanted to use the Coolant output to drive the Air assist solenoid.   Noted now!

openhardwarecoza, 2016-01-21:
https://github.com/openhardwarecoza/LaserWeb/commit/2b92d9478b54ce44cead7c2a7d778336fba1f580  - fix for the air assist buttons - check if it triggers your coolant pin on grbl correctly?

https://github.com/openhardwarecoza/LaserWeb/commit/30f59c74b968439454a91e1029d01f144313be23 covers the new Machine Profiles work in progress:
Especially, see how I have a JSON field for laseron, laseroff, laser0 and laser100 https://github.com/openhardwarecoza/LaserWeb/blob/master/machines

The idea is once this is working you can say:
My ON command is M3 (or M106 - but m106 jumps the queue in Repetier so be aware)
My 100% is S10000 (10000 in the laser100 field)
My 0% = S100 for example.

cojarbi, 2016-01-21:
Great

openhardwarecoza, 2016-01-21:
Yeah, that way I can push defaults to new users, but as needed you can change to what suits you

openhardwarecoza, 2016-01-21:
And it will be saving it per machine - so if you have a co2 and a diode, you can have two profiles, one witch speaks smoothie and one that speaks grbl for example.  Sad that the firmware makers dont collaborate and standardise a little :(

Mad-Onion, 2016-01-21:
@openhardwarecoza the "Zero Machine" Button is still missing in Machine Control for grbl.
But thanks for the awesome other features you added in the last hours :)

openhardwarecoza, 2016-01-22:
Whats the correct gcode to send to zero on grbl?  Also g92 x0 y0 z0?

Mad-Onion, 2016-01-22:
That's correct!

openhardwarecoza, 2016-01-22:
Pushed. Test and revert please

Mad-Onion, 2016-01-22:
Now the Fan-Speed-buttongroup showed up (was also missing before), but no Zero-Machine-button.

openhardwarecoza, 2016-01-22:
Multitasking - lost concentration. Try now

Mad-Onion, 2016-01-22:
Works like a charm :)

openhardwarecoza, 2016-01-22:
Fan too?  (Does grbl even have a fan out?)

Mad-Onion, 2016-01-22:
It has a "coolant enable" pin - but afaik it's not PWM-based.

openhardwarecoza, 2016-01-22:
Ok will change it to remove fan button.  I am using the coolant pin as Air Assist (to fire a solenoid valve) with m8/m9 already

openhardwarecoza, 2016-01-22:
Fan removed from GRBL

commit 2b92d947 (2016-01-21): Added Air On and Off functions to fix #74
commit ba5fe866 (2016-01-21): Merge branch 'master' of https://github.com/fablab-ka/LaserWeb

* 'master' of https://github.com/fablab-ka/LaserWeb:
  Added Grbl M3/m5 to svg2gcode for #73
  Added Grbl M3/m5 to svg2gcode for #73
  Added Air On and Off functions to fix #74
  Basic framework for #7
commit efda3e1c (2016-01-22): Removed Fan for fixed #74
closed 2016-01-22 by openhardwarecoza