LaserWeb/deprecated-LaserWeb1 #77: Lasersaur Machine Control not working after Git pull

tsanalytics, 2016-01-21:
Just did a git pull and now machine control no longer works. I can only home and have to Clear Q after I try and use the X and Y control.

mostley, 2016-01-21:
Probably related to my commit then, weird because I just used it with an lasersaur 30mins ago. What did you put into your config.js and so on?

tsanalytics, 2016-01-21:
I only changed the baud rate, the rest is standard. It even hangs up when I try and draw a bounding box

mostley, 2016-01-21:
That should have nothing to do with my commit then. To get it working with Lasaur I added the laser bounds, the baudrate and these pre- and postamble settings:

```javascript
config.gcodePreamble = [
  'G90',
  'G0 Z 90',
  'M82'
];

config.gcodePostamble = [
  'M83',
  'G0 X0 Y0'
];
```

mostley, 2016-01-21:
oh and I also had an issue with the Lasaur firmware being recognized since it used the ? for keepalive and initialization. I extended my firmware to react to "version". Can you try Homing what command is it trying to execute? G28 or G30?
if G28 try this: connect lasaur, start laserweb, reset lasaur arduino, connect laserweb (via website)

tsanalytics, 2016-01-21:
I just realized that the chiller output has to be true to execute the mentioned functions. Had my pump off while playing with the head position. All seems to be fine now, thanks

tsanalytics, 2016-01-21:
Just tried cutting a DXF file and noticed that it pauses after every move. I have to pause and unpause for it to move again. Totally different issue though.

closed 2016-01-22 by tsanalytics

openhardwarecoza, 2016-01-22:
That pause you see might be a z move (you machine doesn't have a z so it looks like a pause).   Set material thickness to 0.00001 and depth per pass to same