LaserWeb/deprecated-LaserWeb1 #82: Bug in raster with smoothie specific gcode

openhardwarecoza, 2016-01-22:
@cojarbi has an issue on smoothies specificly. 

@cojarbi, please add screenshots, result photos, saved gcode, etc that would help us see what it is

cojarbi, 2016-01-22:
Bellow youll see picture and gcode for the default setting in LW when connect to smoothie. Also for comparison the standard render without being conected, same default settings.
[image]
[image]

[GCODE-LaserWeb Default Settings Smoothie.gcode.txt](https://github.com/openhardwarecoza/LaserWeb/files/101273/GCODE-LaserWeb.Default.Settings.Smoothie.gcode.txt)

cojarbi, 2016-01-22:
Here is a picture of some test done yesterday.  All of them showed bad renders as above. Different setting in speed and power were applied. 
[image]

openhardwarecoza, 2016-01-22:
Pull and test please

cojarbi, 2016-01-22:
Same result on render

openhardwarecoza, 2016-01-22:
I can generate my own gcode, and I havent fixed the render. Test on machine  (what you see isnt always what you get.  The code contains 0.1, 0.2,0.3, -1.0 all of it.  How does it look on your burn

openhardwarecoza, 2016-01-22:
With smoothie:
[image]

openhardwarecoza, 2016-01-22:
no wait, sorry ignore that last commit from fix viewer

openhardwarecoza, 2016-01-22:
I'm tired enough that I am chasing my tail...

cojarbi, 2016-01-22:
I'm lost then.

openhardwarecoza, 2016-01-22:
Found it!

closed 2016-01-22 by openhardwarecoza

openhardwarecoza, 2016-01-22:
@nathanielstenzel  - i figure you may appreciate the discovery of the issue:

1.  The gcode and viewer code BOTH had mistakes:

1.1  The Gcode writer was using map 0-100 -> 0.1 (which doesnt do decimals - which you had identified already) - Fixed in https://github.com/openhardwarecoza/LaserWeb/commit/73291423f915c4ee986a8316bb8206b7364ec54f
1.2  The 3D Viewer had a math.round which was taking the decimals in smoothie gcode and rounding it to either 1 or 0! (so 0.1, 0.2, 0.3 all became 0 (white) and 0.5, 0.6, 0.7 etc all became 1.0 (black) - Fixed in https://github.com/openhardwarecoza/LaserWeb/commit/a6e67ac92dc4a519068b42bfa392abb9b12d9456

nathanielstenzel, 2016-01-22:
Yes, I very much appreciate that fix in the other location. I was working with Ariel and only getting anywhere with speed adjustments of which increasing speed decreased reliability of reaching the required burn for light portions. That should make Ariel's next test go right. If it does, there is probably no need to do the adjustment that I was thinking of for making the white speed lower than black speed when desired. Of course, some of that testing was on gshield, but that is another story with another load of bugs and incompatibilities to consider.

commit a6e67ac9 (2016-01-22): There we go! The math.round was making it either 1 or 0.  Fixed #82