LaserWeb/deprecated-LaserWeb1 #83: Include LibLaserCut for more devices

t-oster, 2016-01-22:
Hi,
I am the creator of VisiCut (https://github.com/t-oster/VisiCut), which is a java based desktop application which does pretty much the same thing.
I always wanted to go webbased, but I do not have mucht time for development anymore.

I just stumbeled over this project and it looks very impressing. Maybe you are interested in sharing a common structure for Lasercutter "drivers", so the software can support more lasercutters over time.

I divided VisiCut into the GUI and a separate libraray (https://github.com/t-oster/LibLaserCut) where the drivers live. It may be possible to use the library in nodejs via the node-java module (http://stackoverflow.com/questions/4729013/can-i-call-java-from-node-js-via-jni-and-how/9352818#9352818) and depending on how you control lasercutters it may be not so hard to include it.

Just a suggestion, since as already mentioned I do not have much time.

openhardwarecoza, 2016-01-22:
Hi.   @quillford (who wrote your Marlin driver)  also suggested this before.  I'll leave this open.  Definitely something to consider.  Always open for pulls and other efforts,  so feel free to redirect some devs over (:

openhardwarecoza, 2016-01-22:
Other little known fact,  but i have been down the visicut route:.  See https://github.com/t-oster/VisiCut/issues/310 -  closed the issue when i gave up.  Was one of the driving factors for starting LaserWeb (;

closed 2016-02-28 by openhardwarecoza