LaserWeb/deprecated-LaserWeb1 #85: Errors in GRBL 9j (jtech)

cojarbi, 2016-01-24:
First some errors on the terminal. I dont think i saw them yersterday. See attached

Also random vertical movements in the middle of lasering. I could not replicated back to back the same movement but in all test the is a vertical one. Attached gcode

At the end of the run the laser stays On. Only way to shut it down is first M03 and then M05
[grbl error terminal LW.txt](https://github.com/openhardwarecoza/LaserWeb/files/102582/grbl.error.terminal.LW.txt)
[grbl jtech GCODE-LaserWeb (1).gcode.txt](https://github.com/openhardwarecoza/LaserWeb/files/102583/grbl.jtech.GCODE-LaserWeb.1.gcode.txt)

cojarbi, 2016-01-24:
Here is another gcode. This one seems to skip dots in the same line and then does them on the return.
[GRBL skips dots.gcode.TXT](https://github.com/openhardwarecoza/LaserWeb/files/102602/GRBL.skips.dots.gcode.TXT)

cojarbi, 2016-01-24:
Another one. Long Y movement towerds the end of the code. Also a lot of hiccups during line codes ( stops for less than a second )
[LADY long vertical movement GCODE-LaserWeb (1).gcode.txt](https://github.com/openhardwarecoza/LaserWeb/files/102607/LADY.long.vertical.movement.GCODE-LaserWeb.1.gcode.txt)

openhardwarecoza, 2016-01-25:
@nathanielstenzel

funinthefalls, 2016-01-25:
I wondered where you went, I am used to seeing you spending tons more time working on this project. Don't worry about the schematic for my project, I will buckle down and relearn how to use eagle. While you are busy at work I will try to finish up the Wiki for LaserWeb, I am about 95% complete now.

nathanielstenzel, 2016-01-26:
When did this start happening and when was your last update before that?

cojarbi, 2016-01-26:
I started with GRBL yesterday so I think there have been 2 commits after that.  It's started since I connected yesterday. 

Ariel Yahni

openhardwarecoza, 2016-01-26:
@funinthefalls yeah that festive slowdown is over!  Work sits in piles in my desk :(

cojarbi, 2016-01-27:
I have made a mistake in the title, so all this issues are for 0.9g modified by Jtech

openhardwarecoza, 2016-01-27:
#88  - see Stock Grbl (;

cojarbi, 2016-01-27:
Stock grbl. 9j uses power range 0-10000

athulsnair, 2016-01-28:
@cojarbi not only that , after reading through shapeoko forum this is what I got , grbl flushes the buffer before changing spindle speed. So when there is a spindle speed change , grbl assumes that you are using a milling machine and it will not accelerate (it stops ) so that spindle can achieve new speed i.e., machine will not accelerate smoothly through spindle speed changes. 
In modified versions this is avoided.

openhardwarecoza, 2016-01-28:
Right, so while I was at work, looks like you two made some progress.

Do we need to add the m3/m5 to raster still?  Or is grbl now honoring the power level change as enough (the less serial data we need to send  the better)

More importantly - we need to finish #7 - then you can configure your own machine profile from the UI (go see screenshot at bottom of the thread on issue no 7 I posted a few mins ago) without any coding in the backend...

cojarbi, 2016-01-29:
Today i flashed 0.9j master and modified it to S255. Since laserweb is not seding M03 commands the laser does not turn on

athulsnair, 2016-01-29:
@cojarbi Edit the gcode(add m3 at the beining and m5 at the end) that's what i'm doing

athulsnair, 2016-01-29:
@openhardwarecoza  grbl 0.9j requires m3 and m5

openhardwarecoza, 2016-01-29:
I am simply drowing in work

Open laserraster.js and add the code to each move

Send me a pull request after testing...

openhardwarecoza, 2016-01-29:
In https://github.com/openhardwarecoza/LaserWeb/blob/master/i/raster/laserraster.js#L223

```
if (this.config.useVariableSpeed) {
          this.result += 'M3\n';
          this.result += 'G1 X{0} Y{1} S{2} F{3}\n'.format(posx, gcodey, lastIntensity, speed);
          this.result += 'M5\n';
        } else {
          this.result += 'M3\n';
          this.result += 'G1 X{0} Y{1} S{2}\n'.format(posx, gcodey, lastIntensity);
          this.result += 'M5\n';
        }
```

Change to 

```
if (this.config.useVariableSpeed) {
          this.result += 'G1 X{0} Y{1} S{2} F{3}\n'.format(posx, gcodey, lastIntensity, speed);
        } else {
          this.result += 'G1 X{0} Y{1} S{2}\n'.format(posx, gcodey, lastIntensity);
        }
```

openhardwarecoza, 2016-01-29:
Pushed - do a git pull update and let me know!

athulsnair, 2016-01-29:
There're m3 and  m5 commands and laser output, but noticed several unwanted movements (even though machine comes back to the required position) and there was a long pause at a time (I didn't do full engraving, so don't know if there's more than one).

I also like to mention one more thing,I generated a raster gcode using laserweb , dowloaded the genrated gcode, manually inserted M3 and M5 command and run it by laserweb, later I run the same gcode using chilipeppr and result was quite different from what I got from laserweb.
In laserweb there was lots of gaps ( there's wasn't laser sometimes) and error messages were
   invalid gcode ID:24 and
   bad number format

Top one in the image is from chilipeppr and bottom one from laserweb
[image]

I'm also attaching modified gcode(manually inserted m3 and m5) and gcode I got after new fix (raster with M3 and M5) and png image
[manually inserted M3 and M5.docx](https://github.com/openhardwarecoza/LaserWeb/files/109741/manually.inserted.M3.and.M5.docx)
[raster with m3 and m5.docx](https://github.com/openhardwarecoza/LaserWeb/files/109742/raster.with.m3.and.m5.docx)
[image]

openhardwarecoza, 2016-01-29:
Was SPJS shutdown while you were testing with Laserweb? Those unknown gcode messages pop up when something else talks to the controller at the same time (; 

ID:24 doesnt appear in your gcode nor in grbls responses, so looks like something else was talking to that port too

openhardwarecoza, 2016-01-29:
As a sidenote, I wouldnt mind someone working on https://github.com/openhardwarecoza/LaserWeb/issues/21 - so if you feel the serial comms could be better, open a seperate thread (if if we have to use SPJS too (;  -)

openhardwarecoza, 2016-01-29:
Mind you, looking back at post one, cojarbi also had a ID2x error in his log... very weird, where is it coming from?

openhardwarecoza, 2016-01-29:
Ok, let me guess. you are both on Mac, and instead of seeing the nicely colored lines, your crapware chokes on the ascii code and prints things like ID24 instead of coloring it :/

It should look like this:  
[image]

And yes yes, I know I shouldntt hate on mac.,  But come on.... lol

openhardwarecoza, 2016-01-29:
Gents, fix your Macs:  See http://stackoverflow.com/questions/29364326/how-to-use-chalk-terminal-string-styling-done-right so we can get better logs out thats actually usable

openhardwarecoza, 2016-01-29:
Continuing my one sided rant:

Your error says it complains about Number Format
Check if the S values match what grbl expects?  (decimals etc?)

openhardwarecoza, 2016-01-29:
The docx's you uploaded (Why, theres a big SAVE GCODE button?) shows the m3/m5s in the right spots now.  So i think (to avoid convulated consufing issue threads) we close this one and handle the serial issues (since the exported file runs fine in another host) seperately.  I'll open a new issue with debug instructions
Since thats not grbl specific... (Chirag had a similar issue with Marlin)

closed 2016-01-29 by openhardwarecoza

cojarbi, 2016-01-29:
@athulsnair and I have something else common I believe. We bough tried chillippeper and now I think that the json serial server could be the issue. I did went over their git asking how to uninstall it completely but the response was that if it's off then should not interfere with anything

openhardwarecoza, 2016-01-29:
kill the process... then its gone (delete the executable if you are scared it is running)

cojarbi, 2016-01-29:
I killed the MF with a laser :) 

Just thinking what you said about something else talking in there

Ariel Yahni

athulsnair, 2016-01-30:
@openhardwarecoza  I'm windows 10 user not MAC           

And I don't know anything about coding .I'm trying to modify my X carve in to  a laser cutter and testing it by using laser web.
Are you saying JSON server that I use for the chilipeppr is communicating with the serial port ?? and that's why I'm getting all the errors??

openhardwarecoza, 2016-01-30:
Its possible,  but refer the other issue.. We have a few prople working on this now.  Sit tight for a day or two,  we'll fix the serial bug soon