LaserWeb/deprecated-LaserWeb1 #87: Support for Wifi SD cards (Flashair, etc)

openhardwarecoza, 2016-01-27:
An idea from John over on the Hackaday Comments Threads:
http://hackaday.com/2016/01/26/stop-driving-laser-cutters-with-3d-printer-software/#comment-2898642

Use a Wifi SD installed on the controller.

I think I'll order one and start playing to add a "push to SD" button in Laserweb that uploads it to these wifi SD cards - that would be a win!

openhardwarecoza, 2016-02-09:
https://github.com/openhardwarecoza/LaserWeb/wiki/SPONSORS -  feel like donating some hardware? 
Or if you have more info how we can do the upload using a Ajax POST?

funinthefalls, 2016-02-09:
Let me check my Wanhoa users group, if I remember correctly, someone managed to make this idea work very easily with a 3dprinter. Hopefully I can find the thread.

funinthefalls, 2016-02-09:
https://flashair-developers.com/en/documents/tutorials/advanced/2/

closed 2016-05-11 by openhardwarecoza