LaserWeb/deprecated-LaserWeb1 #88: Laser on while connecting

athulsnair, 2016-01-27:
Laser is ON for a second while starting laserWeb.

openhardwarecoza, 2016-01-27:
If your laser is on during starting it means your hardware has floating pins!   Add pulldown resistors! 

(Give me more details on what hardware and firmware you are using if you need more help... )

Regarding m3/m5 - same - what firmware are you using/ Can't help you out if you dont log detailed issues

athulsnair, 2016-01-27:
I'm using GRBL 0.9j with Gshield (pwm from 0-255)

openhardwarecoza, 2016-01-27:
Ok, so for 0.9J - refer to the duplicate bug already logged: https://github.com/openhardwarecoza/LaserWeb/issues/85

For the reset issue - add pulldown resistors to gnd on your laser out pins (or keep laser power off until ready to cut)

athulsnair, 2016-01-27:
What I understand from reading that post is it uses modified version of grbl and I'm using original version and I don't have any problem with cutting by uploading a dxf file.(it has m3 and m5 commands and S parameter from 0-255 , that's enough form me).

openhardwarecoza, 2016-01-27:
Ok,  will take a look shortly.

cojarbi, 2016-01-27:
@athulsnair are you sure u are using Master. 9j and it's power range is S0-255? I'm sure I've tried that and it's 0-10000 based on spindle rpm

cojarbi, 2016-01-27:
https://github.com/grbl/grbl/issues/892

athulsnair, 2016-01-28:
I changed the config.h file in grbl master .
look for #define_max_spindle_rpm in config.h file,by default it's 1000rpm, change it to 255 so that S255 comment  will give you 5V on pin 11 (variable spindle should be enabled)

athulsnair, 2016-01-28:
[](https://github.com/grbl/grbl/issues/877#issuecomment-170151182)

openhardwarecoza, 2016-01-29:
M3+m5 added https://github.com/openhardwarecoza/LaserWeb/commit/093233c910843aed856eb4e84078f67ee04b9f31

closed 2016-01-29 by openhardwarecoza