LaserWeb/deprecated-LaserWeb1 #91: Support Rep Rap Firmware & Duet Boards?

BrianGilbert, 2016-01-28:
Just wondering what would be involved in getting support for the Duet 0.8.5 board which runs on RepRapFirmware?

openhardwarecoza, 2016-01-28:
Hi @BrianGilbert 

Unfortunately I don't have on of those on my desk, but you do (:  

Do you feel up for playing a little coding tag team? 

Adding support should be really easy:

## Server.js
1.  Below https://github.com/openhardwarecoza/LaserWeb/blob/master/server.js#L174 add a write for whatever command we can send to the duet in order for it to report what it is running (see the examples between lines 174-176)

2.  Add on https://github.com/openhardwarecoza/LaserWeb/blob/master/server.js#L308,   a copy pasted block of the same code as lines 292->307 (https://github.com/openhardwarecoza/LaserWeb/blob/master/server.js#L292) or same as 256-269, or 271-289 (enough examples of the existing firmwares) to tell the frontend about the Duet/Reprapfirmware

3.  Format the position and temperature feedback
Example code for this is around https://github.com/openhardwarecoza/LaserWeb/blob/master/server.js#L311 downward to around line 382 that you can copy paste if needed.  (may not need to add anything if the duet firmware emits a response that matches what we look for there, for example https://github.com/openhardwarecoza/LaserWeb/blob/master/server.js#L319 - but look over the block, even between Repetier and Marlin I had to catch both seperately

The firmware value is then sent to the UI on websocket firstload

##main.js
Once main.js receives the firmware at 
https://github.com/openhardwarecoza/LaserWeb/blob/master/i/main.js#L2167
you'll need to add jquery (use the examples down from line 2167 to 2228 to customise which buttons show up in the UI and which doesnt 

Then all over there are some if/else going on, for example https://github.com/openhardwarecoza/LaserWeb/blob/master/i/main.js#L1151 to handle slight differences between what each firmware expects (The linked example here is home commands.  Reprap type firmwares want G28, Lasaurgrbl wanted ~\nG30 and Grbl wants a $H -   so any case where there differences in commands that Duet specifically want you can add if/else like this

Then in the gcode writers, we do the same:  
For example Smoothieware wants the Sxxx parameter between S0 and S1 (for example 50% laser power is S0.5) but LasaurGrbl wants S0-255 (So 50% power is S128)
To accomodate this, you can look at:
https://github.com/openhardwarecoza/LaserWeb/blob/master/i/svg/js/svg2gcode.js#L182
https://github.com/openhardwarecoza/LaserWeb/blob/master/i/raster/laserraster.js#L115
https://github.com/openhardwarecoza/LaserWeb/blob/master/i/millcrum/mc.js#L759
https://github.com/openhardwarecoza/LaserWeb/blob/master/i/gcode-viewer/gcode-parser.js#L372  (Why in the viewer, otherwise we render lines darker or lighter than they should be

My suggestion is fork, create a development branch and start working on it. No one is more capable of adding it than you, since you hold the key piece of hardware in hand to do rapid testing and debugging (:
Once we have it working we do a pull request (:   (Or I can give you push access - always happy for more contributors)

Lets keep this thread running, I'm on here maybe 5-6 times per day, and if its really urgent, tag me into a post on Google Plus https://plus.google.com/+PetervanderWalt

Sorry if the post is dumbed down - but I think I can use this going forward as a good template to post when the next person wants to add an extra firmware 

Something else worth mentioning, I want to move these firmware specific gcodes our of the if/else way of doing it.  See #7   - the idea is that we can set these (I need a M3 to start, and you dont.  I need S0-1, and you need S0-255, and you need S0-10000, etc etc) as variables in the config of your "machine profile"  - should make adding more firmware (or version differences) easier - since you can configure it from the frontend instead of having to change code in 5 different spots...

openhardwarecoza, 2016-01-28:
Ohg, and remember once you have some code merged,you are welcome to add your paypal link / bio to https://github.com/openhardwarecoza/LaserWeb/wiki/SPONSORS - we actually let the community support the contributors! Without good contributors, there would not be a project

openhardwarecoza, 2016-05-11:
Closing this one.  See github.com/openhardwarecoza/LaserWeb2 instead

closed 2016-05-11 by openhardwarecoza