LaserWeb/deprecated-LaserWeb1 #93: adding Rotary

Pensee, 2016-01-29:
Stephane BuissonOWNER10:52 AM
+Peter van der Walt an option for Laserweb maybe ...﻿

Peter van der WaltMODERATOR10:58 AM
Add a github issue so we remember to work on it. Baseline y dimension, scaled along difference between baseline diameter and item being engraved diameter, while keeping the existing x scaling. Can use my scaling gcode functions as is (; just need to wrap in a UI. The thing with rotaries is to Remember different diameters act like different steps per mm. Thats easy﻿

openhardwarecoza, 2016-01-29:
Support for rotary axis

funinthefalls, 2016-01-30:
That would be great Pter. I have a few differnt rotary designs I would like to try out.

openhardwarecoza, 2016-05-11:
I still have this in mind, but closing this issue to reopen on LaserWeb2

closed 2016-05-11 by openhardwarecoza