LaserWeb/deprecated-LaserWeb1 #96: Question : Smoothie board ignored

Protoneer, 2016-01-30:
I am trying to use LaserWeb on a RPI B+ (V1) and I am not able to connect to my SmoothieBoard. Console comes up correctly but returns the following message:

```
WARN: Ignored:  Port /dev/ttyUSB0 said:  Build version, edge-2163ccd, Build date, Dec 28 2015 15,22,39, MCU, LPC1769, System Clock, 120MHz
problem reading Machine Profiles, using none

```

Does this point to Smoothie not being configured correctly or is this on the the RPI's side?

Protoneer, 2016-01-30:
With the above issue there is no serial ports in LaserWeb to connect to.

cojarbi, 2016-01-30:
Does it connect to your computer directly?

Protoneer, 2016-01-30:
Yes... if I open a terminal app like Minicom it works normal.

Protoneer, 2016-01-30:
It does not look like a coms error cause the message includes the version
details from Smoothie...

cojarbi, 2016-01-30:
Thus you discarded smoothie since you are able to control your machine locally I assume. Then is on the RPI side. Did you see the incompatibilities of serial port on node version

Protoneer, 2016-01-30:
I am running Node V0.12.6 and Serial 1.6.1 on Jesse. I could not get Serial v2+ to even install on the RPI. I will try the setup with GRBL shortly...

Protoneer, 2016-01-30:
Mmmmm.... that's also not happy with GRBL.

I enabled "Enable Serial Receive debug".

```
INFO: adding /dev/ttyAMA0 because it is enabled in config.js, you may need to enable it in the os - http://www.hobbytronics.co.uk/raspberry-pi-serial-port
ERROR: Error connecting to webcam: connect ECONNREFUSED - Disabling Webcam Widget
Port 0 got newline from serial: <Idle,MPos:0.000,0.000,0.000,WPos:0.000,0.000,0.000>
Port 0 got newline from serial: ok
Port 0 got newline from serial: error: Unsupported command
ERROR:  Error from machine: error: Unsupported command
Aborted Job - safety first!
Port 0 got newline from serial: error: Bad number format
ERROR:  Error from machine: error: Bad number format
Aborted Job - safety first!
problem reading Machine Profiles, using none
```
Looking at the line 4 of the above I think the newline is causing an extra OK [https://github.com/openhardwarecoza/LaserWeb/blob/master/server.js#L242](https://github.com/openhardwarecoza/LaserWeb/blob/master/server.js#L242)

openhardwarecoza, 2016-01-31:
Do a test with running the queryloop without the newline please (see if you still get DRO updates?)

Regarding Smoothie, if you arent getting any ports in the list, its Node Serialport

If you do get ports, but its not identifying the firmware, or showing the firmware version on the button above the serial dropdown when you select the dropdown - then its laserweb

Tiaan (just up road from here) runs on a Pi 2 with the correct version of Serialport as per the instuctions and she goes...

openhardwarecoza, 2016-02-09:
Any feedback?

openhardwarecoza, 2016-02-10:
No feedback

closed 2016-02-10 by openhardwarecoza