LaserWeb/deprecated-LaserWeb1 #97: Cannot connect to Marlin based board

mrgastoso, 2016-01-31:
Hi, im testing with an Marlin Turnkey firmware laser, but seems that the port is ignored, the board is an China clone with the HG-340 serial chip, not sure if this will be a problem. 

***************************************************************
*                        Notice:                              *
***************************************************************
*     Remember to update (: !!!                               *
* 1.  Run ./update.sh or git pull                             *
* 2.  or check the commit log on                              *
* https://github.com/openhardwarecoza/LaserWeb/commits/master *
***************************************************************
* Access the LaserWeb User Interface:                         *
* 1. Open Chrome                                              *
* 2. Go to :                                                  *
*    http://127.0.1.1:8000/                                   *
***************************************************************
 
 
ERROR: Error connecting to webcam: connect ECONNREFUSED 127.0.0.1:8080 - Disabling Webcam Widget
WARN: Ignored:  Port /dev/ttyUSB0 said:  start
INFO:  Found device:  1a86 /dev/ttyUSB0 Firmware Detected: Marlin 1.0.0
WARN: Ignored:  Port /dev/ttyUSB0 said:  echo:Marlin 1.0.0
WARN: Ignored:  Port /dev/ttyUSB0 said:  echo: Last Updated: Jan 31 2016 11:31:23 | Author: (none, default config)
WARN: Ignored:  Port /dev/ttyUSB0 said:  Compiled: Jan 31 2016
WARN: Ignored:  Port /dev/ttyUSB0 said:  echo: Free Memory: 1910  PlannerBufferBytes: 2576
WARN: Ignored:  Port /dev/ttyUSB0 said:  echo:Stored settings retrieved
WARN: Ignored:  Port /dev/ttyUSB0 said:  echo:Steps per unit:
WARN: Ignored:  Port /dev/ttyUSB0 said:  echo:  M92 X40.00 Y40.00 Z1000.00 E0.00
WARN: Ignored:  Port /dev/ttyUSB0 said:  echo:Maximum feedrates (mm/s):
WARN: Ignored:  Port /dev/ttyUSB0 said:  echo:  M203 X2000.00 Y1000.00 Z10000.00 E0.00
WARN: Ignored:  Port /dev/ttyUSB0 said:  echo:Maximum Acceleration (mm/s2):
WARN: Ignored:  Port /dev/ttyUSB0 said:  echo:  M201 X2600 Y60 Z10000 E2
WARN: Ignored:  Port /dev/ttyUSB0 said:  echo:Acceleration: S=acceleration, T=retract acceleration
WARN: Ignored:  Port /dev/ttyUSB0 said:  echo:  M204 S3000.00 T3000.00
WARN: Ignored:  Port /dev/ttyUSB0 said:  echo:Advanced variables: S=Min feedrate (mm/s), T=Min travel feedrate (mm/s), B=minimum segment time (ms), X=maximum XY jerk (mm/s),  Z=maximum Z jerk (mm/s),  E=maximum E jerk (mm/s)
WARN: Ignored:  Port /dev/ttyUSB0 said:  echo:  M205 S0.00 T0.00 B0 X20.00 Z0.40 E0.00
WARN: Ignored:  Port /dev/ttyUSB0 said:  echo:Home offset (mm):
WARN: Ignored:  Port /dev/ttyUSB0 said:  echo:  M206 X-5.00 Y-5.00 Z0.00
WARN: Ignored:  Port /dev/ttyUSB0 said:  echo:Laser lifetime usage:
WARN: Ignored:  Port /dev/ttyUSB0 said:  echo: Hours: 0
WARN: Ignored:  Port /dev/ttyUSB0 said:   Minutes: 0
WARN: Ignored:  Port /dev/ttyUSB0 said:  echo:PID settings:
WARN: Ignored:  Port /dev/ttyUSB0 said:  echo:   M301 P22.20 I1.08 D114.00
INFO: Machine Profiles read from file
Missing error handler on `socket`.
TypeError: Cannot read property 'port' of undefined
    at Socket.<anonymous> (/tmp/LaserWeb/server.js:748:26)
    at emitOne (events.js:90:13)
    at Socket.emit (events.js:182:7)
    at Socket.onevent (/tmp/LaserWeb/node_modules/socket.io/lib/socket.js:330:8)
    at Socket.onpacket (/tmp/LaserWeb/node_modules/socket.io/lib/socket.js:290:12)
    at Client.ondecoded (/tmp/LaserWeb/node_modules/socket.io/lib/client.js:193:14)
    at Decoder.Emitter.emit (/tmp/LaserWeb/node_modules/component-emitter/index.js:134:20)
    at Decoder.add (/tmp/LaserWeb/node_modules/socket.io-parser/index.js:247:12)
    at Client.ondata (/tmp/LaserWeb/node_modules/socket.io/lib/client.js:175:18)
    at emitOne (events.js:90:13)
WARN: Ignored:  Port /dev/ttyUSB0 said:  echo:SD card ok

nathanielstenzel, 2016-01-31:
I have no experience wifh board interfacing, but I wonder if " Author: ( none, default config)" has something to do with it. It just popped out at me when reading this.
On a side note, I have seen a few other items on the issues list where they were trying to imprive the buffer and support for certain boards. That work could also be relevent.
The people who workon the board support will propably need to know if you can connect to the board using other software or not.

openhardwarecoza, 2016-01-31:
The CH340 shouldn't be a problem (i actually started to like the ch340 the last few months and use them on my own PCBs quite a bit.   That said, I actually havent used on against Laserweb (or more specifically node.js SerialPort) yet...

Line 748, deals wth logging info when you select a port in the UI:
Just tested now with a MJDuino clone I have lying around here, running Turnkey:
Here's what my start looks like:
ERROR: Error connecting to webcam: connect ECONNREFUSED - Disabling Webcam Widget
WARN: Ignored:  Port /dev/ttyACM0 said:  start
INFO:  Found device:  Arduino__www.arduino.cc_ /dev/ttyACM0 Firmware Detected: Marlin 1.0.0
WARN: Ignored:  Port /dev/ttyACM0 said:  echo:Marlin 1.0.0
WARN: Ignored:  Port /dev/ttyACM0 said:  echo: Last Updated: Nov 10 2015 10:51:58 | Author: (Peter vd Walt)
WARN: Ignored:  Port /dev/ttyACM0 said:  Compiled: Nov 10 2015
WARN: Ignored:  Port /dev/ttyACM0 said:  echo: Free Memory: 2540  PlannerBufferBytes: 2576
WARN: Ignored:  Port /dev/ttyACM0 said:  echo:Hardcoded Default Settings Loaded
WARN: Ignored:  Port /dev/ttyACM0 said:  echo:Steps per unit:
WARN: Ignored:  Port /dev/ttyACM0 said:  echo:  M92 X53.50 Y53.50 Z3260.00 E5000.00
WARN: Ignored:  Port /dev/ttyACM0 said:  echo:Maximum feedrates (mm/s):
WARN: Ignored:  Port /dev/ttyACM0 said:  echo:  M203 X5000.00 Y5000.00 Z5000.00 E5000.00
WARN: Ignored:  Port /dev/ttyACM0 said:  echo:Maximum Acceleration (mm/s2):
WARN: Ignored:  Port /dev/ttyACM0 said:  echo:  M201 X9000 Y9000 Z9000 E9000
WARN: Ignored:  Port /dev/ttyACM0 said:  echo:Acceleration: S=acceleration, T=retract acceleration
WARN: Ignored:  Port /dev/ttyACM0 said:  echo:  M204 S8000.00 T8000.00
WARN: Ignored:  Port /dev/ttyACM0 said:  echo:Advanced variables: S=Min feedrate (mm/s), T=Min travel feedrate (mm/s), B=minimum segment time (ms), X=maximum XY jerk (mm/s),  Z=maximum Z jerk (mm/s),  E=maximum E jerk (mm/s)
WARN: Ignored:  Port /dev/ttyACM0 said:  echo:  M205 S0.00 T0.00 B1 X20.00 Z20.00 E5.00
WARN: Ignored:  Port /dev/ttyACM0 said:  echo:Home offset (mm):
WARN: Ignored:  Port /dev/ttyACM0 said:  echo:  M206 X0.00 Y0.00 Z0.00
WARN: Ignored:  Port /dev/ttyACM0 said:  echo:Laser lifetime usage:
WARN: Ignored:  Port /dev/ttyACM0 said:  echo: Hours: 0
WARN: Ignored:  Port /dev/ttyACM0 said:   Minutes: 0
WARN: Ignored:  Port /dev/ttyACM0 said:  echo:PID settings:
WARN: Ignored:  Port /dev/ttyACM0 said:  echo:   M301 P22.20 I1.08 D114.00
INFO: Machine Profiles read from file
INFO:  Now interacting with  /dev/ttyACM0  running  Marlin 1.0.0
Checking for Updates
INFO: Git Refresh Fetching origin

So yours got to the machine profiles too.  The next step is to hang tight till you select a dropdown in the UI
Once a port is selected, we log out to screen which port that was. 

So, that said, it might be serial port related with the ch340...

Quick mental checks:
1. You can access the port from say pronterface? (i.e your posix user account has the right tty//dialout groups?)
2.  Board was plugged in before you started Laserweb?
3.  a ctrl+c and start again tried?
4.  sudo dmesg -c before starting laserweb, replicate error, then  'dmesg'

nathanielstenzel, 2016-01-31:
If you can't make Marlin gcode either, there is a debug mode for when you run "node server.js" but I forget the details at this time. If other software can connect, you could then save the marlin gcode and load it into the other softward as a momentary work around.

openhardwarecoza, 2016-01-31:
@nathanielstenzel  you advice is way off and just going to confuse the user...

openhardwarecoza, 2016-01-31:
@mrgastoso - i added a extra logging for us, to see whether the CH340 does something different in the array, please 'git pull' then test again, paste new log

mrgastoso, 2016-01-31:
I have tested again and this is the results:

ERROR: Error connecting to webcam: connect ECONNREFUSED 127.0.0.1:8080 - Disabling Webcam Widget
WARN: Ignored:  Port /dev/ttyUSB0 said:  start
INFO:  Found device:  1a86 /dev/ttyUSB0 Firmware Detected: Marlin 1.0.0   Port ID No:  32
WARN: Ignored:  Port /dev/ttyUSB0 said:  echo:Marlin 1.0.0
WARN: Ignored:  Port /dev/ttyUSB0 said:  echo: Last Updated: Jan 31 2016 11:31:23 | Author: (none, default config)
WARN: Ignored:  Port /dev/ttyUSB0 said:  Compiled: Jan 31 2016
WARN: Ignored:  Port /dev/ttyUSB0 said:  echo: Free Memory: 1910  PlannerBufferBytes: 2576
WARN: Ignored:  Port /dev/ttyUSB0 said:  echo:Stored settings retrieved
WARN: Ignored:  Port /dev/ttyUSB0 said:  echo:Steps per unit:
WARN: Ignored:  Port /dev/ttyUSB0 said:  echo:  M92 X40.00 Y40.00 Z1000.00 E0.00
WARN: Ignored:  Port /dev/ttyUSB0 said:  echo:Maximum feedrates (mm/s):
WARN: Ignored:  Port /dev/ttyUSB0 said:  echo:  M203 X2000.00 Y1000.00 Z10000.00 E0.00
WARN: Ignored:  Port /dev/ttyUSB0 said:  echo:Maximum Acceleration (mm/s2):
WARN: Ignored:  Port /dev/ttyUSB0 said:  echo:  M201 X2600 Y60 Z10000 E2
WARN: Ignored:  Port /dev/ttyUSB0 said:  echo:Acceleration: S=acceleration, T=retract acceleration
WARN: Ignored:  Port /dev/ttyUSB0 said:  echo:  M204 S3000.00 T3000.00
WARN: Ignored:  Port /dev/ttyUSB0 said:  echo:Advanced variables: S=Min feedrate (mm/s), T=Min travel feedrate (mm/s), B=minimum segment time (ms), X=maximum XY jerk (mm/s),  Z=maximum Z jerk (mm/s),  E=maximum E jerk (mm/s)
WARN: Ignored:  Port /dev/ttyUSB0 said:  echo:  M205 S0.00 T0.00 B0 X20.00 Z0.40 E0.00
WARN: Ignored:  Port /dev/ttyUSB0 said:  echo:Home offset (mm):
WARN: Ignored:  Port /dev/ttyUSB0 said:  echo:  M206 X-5.00 Y-5.00 Z0.00
WARN: Ignored:  Port /dev/ttyUSB0 said:  echo:Laser lifetime usage:
WARN: Ignored:  Port /dev/ttyUSB0 said:  echo: Hours: 0
WARN: Ignored:  Port /dev/ttyUSB0 said:   Minutes: 0
WARN: Ignored:  Port /dev/ttyUSB0 said:  echo:PID settings:
WARN: Ignored:  Port /dev/ttyUSB0 said:  echo:   M301 P22.20 I1.08 D114.00
INFO: Machine Profiles read from file
WARN: Ignored:  Port /dev/ttyUSB0 said:  echo:SD card ok
Missing error handler on `socket`.
TypeError: Cannot read property 'port' of undefined
    at Socket.<anonymous> (/tmp/LaserWeb/server.js:758:26)
    at emitOne (events.js:90:13)
    at Socket.emit (events.js:182:7)
    at Socket.onevent (/tmp/LaserWeb/node_modules/socket.io/lib/socket.js:330:8)
    at Socket.onpacket (/tmp/LaserWeb/node_modules/socket.io/lib/socket.js:290:12)
    at Client.ondecoded (/tmp/LaserWeb/node_modules/socket.io/lib/client.js:193:14)
    at Decoder.Emitter.emit (/tmp/LaserWeb/node_modules/component-emitter/index.js:134:20)
    at Decoder.add (/tmp/LaserWeb/node_modules/socket.io-parser/index.js:247:12)
    at Client.ondata (/tmp/LaserWeb/node_modules/socket.io/lib/client.js:175:18)
    at emitOne (events.js:90:13)

INFO: Machine Profiles read from file
Missing error handler on `socket`.
TypeError: Cannot read property 'port' of undefined
    at Socket.<anonymous> (/tmp/LaserWeb/server.js:758:26)
    at emitOne (events.js:90:13)
    at Socket.emit (events.js:182:7)
    at Socket.onevent (/tmp/LaserWeb/node_modules/socket.io/lib/socket.js:330:8)
    at Socket.onpacket (/tmp/LaserWeb/node_modules/socket.io/lib/socket.js:290:12)
    at Client.ondecoded (/tmp/LaserWeb/node_modules/socket.io/lib/client.js:193:14)
    at Decoder.Emitter.emit (/tmp/LaserWeb/node_modules/component-emitter/index.js:134:20)
    at Decoder.add (/tmp/LaserWeb/node_modules/socket.io-parser/index.js:247:12)
    at Client.ondata (/tmp/LaserWeb/node_modules/socket.io/lib/client.js:175:18)
    at emitOne (events.js:90:13)

openhardwarecoza, 2016-01-31:
Port ID32  is very high!  I am used to  devices 0,1,2 etc.  

ls /dev/tty*    
If yoi have a tonne of ttyS1-ttys3x devices,  try a 
sudo rm /dev/ttyS*  

Then unplug,  replug controller,  and try starting server.js again.

openhardwarecoza, 2016-01-31:
It is something pc specific in your case.  

Os?  (ubuntu,  pi, something else?) 
Browser (Chrome?) 
No firewalls that could be screwing with websockets (the undefined error is we arent getting the port id in the websocket response for usePort (sent from browser to server.js over websocket).  We do get the useport message but not the id

mrgastoso, 2016-01-31:
Ive tried to reboot the server and seems that the portid is the same, no udev rules or special config for the serial port, the /dev/ttyUSB* is created/deleted with plug/unplug actions

Im using ubuntu wily 15.10, with chrome 48.0.2564.97, no firewall and selinux enabled

ERROR: Error connecting to webcam: connect ECONNREFUSED 127.0.0.1:8080 - Disabling Webcam Widget
WARN: Ignored:  Port /dev/ttyUSB0 said:  start
INFO:  Found device:  1a86 /dev/ttyUSB0 Firmware Detected: Marlin 1.0.0   Port ID No:  32
WARN: Ignored:  Port /dev/ttyUSB0 said:  echo:Marlin 1.0.0
WARN: Ignored:  Port /dev/ttyUSB0 said:  echo: Last Updated: Jan 31 2016 11:31:23 | Author: (none, default config)
WARN: Ignored:  Port /dev/ttyUSB0 said:  Compiled: Jan 31 2016
WARN: Ignored:  Port /dev/ttyUSB0 said:  echo: Free Memory: 1910  PlannerBufferBytes: 2576
WARN: Ignored:  Port /dev/ttyUSB0 said:  echo:Stored settings retrieved
WARN: Ignored:  Port /dev/ttyUSB0 said:  echo:Steps per unit:
WARN: Ignored:  Port /dev/ttyUSB0 said:  echo:  M92 X40.00 Y40.00 Z1000.00 E0.00
WARN: Ignored:  Port /dev/ttyUSB0 said:  echo:Maximum feedrates (mm/s):
WARN: Ignored:  Port /dev/ttyUSB0 said:  echo:  M203 X2000.00 Y1000.00 Z10000.00 E0.00
WARN: Ignored:  Port /dev/ttyUSB0 said:  echo:Maximum Acceleration (mm/s2):
WARN: Ignored:  Port /dev/ttyUSB0 said:  echo:  M201 X2600 Y60 Z10000 E2
WARN: Ignored:  Port /dev/ttyUSB0 said:  echo:Acceleration: S=acceleration, T=retract acceleration
WARN: Ignored:  Port /dev/ttyUSB0 said:  echo:  M204 S3000.00 T3000.00
WARN: Ignored:  Port /dev/ttyUSB0 said:  echo:Advanced variables: S=Min feedrate (mm/s), T=Min travel feedrate (mm/s), B=minimum segment time (ms), X=maximum XY jerk (mm/s),  Z=maximum Z jerk (mm/s),  E=maximum E jerk (mm/s)
WARN: Ignored:  Port /dev/ttyUSB0 said:  echo:  M205 S0.00 T0.00 B0 X20.00 Z0.40 E0.00
WARN: Ignored:  Port /dev/ttyUSB0 said:  echo:Home offset (mm):
WARN: Ignored:  Port /dev/ttyUSB0 said:  echo:  M206 X-5.00 Y-5.00 Z0.00
WARN: Ignored:  Port /dev/ttyUSB0 said:  echo:Laser lifetime usage:
WARN: Ignored:  Port /dev/ttyUSB0 said:  echo: Hours: 0
WARN: Ignored:  Port /dev/ttyUSB0 said:   Minutes: 0
WARN: Ignored:  Port /dev/ttyUSB0 said:  echo:PID settings:
WARN: Ignored:  Port /dev/ttyUSB0 said:  echo:   M301 P22.20 I1.08 D114.00
WARN: Ignored:  Port /dev/ttyUSB0 said:  echo:SD card ok
INFO: Machine Profiles read from file
Missing error handler on `socket`.
TypeError: Cannot read property 'port' of undefined
    at Socket.<anonymous> (/home/king/Descargas/LaserWeb/server.js:758:26)
    at emitOne (events.js:90:13)
    at Socket.emit (events.js:182:7)
    at Socket.onevent (/home/king/Descargas/LaserWeb/node_modules/socket.io/lib/socket.js:330:8)
    at Socket.onpacket (/home/king/Descargas/LaserWeb/node_modules/socket.io/lib/socket.js:290:12)
    at Client.ondecoded (/home/king/Descargas/LaserWeb/node_modules/socket.io/lib/client.js:193:14)
    at Decoder.Emitter.emit (/home/king/Descargas/LaserWeb/node_modules/component-emitter/index.js:134:20)
    at Decoder.add (/home/king/Descargas/LaserWeb/node_modules/socket.io-parser/index.js:247:12)
    at Client.ondata (/home/king/Descargas/LaserWeb/node_modules/socket.io/lib/client.js:175:18)
    at emitOne (events.js:90:13)

openhardwarecoza, 2016-01-31:
Node5 doesnt work with node-serialport yet. I run Ubuntu 15.10.  Do you have another machine at hand you can try?

openhardwarecoza, 2016-01-31:
In bed already,  but in the usePort function in server.js,  add a console.log('usePort data was ' +data);  so we can see what the browser sends?

openhardwarecoza, 2016-01-31:
Look. At line 750.  Add the console.log to line 751 for instance

mrgastoso, 2016-01-31:
here are the results:

usePort data was null
Missing error handler on `socket`.
TypeError: Cannot read property 'port' of undefined
    at Socket.<anonymous> (/tmp/LaserWeb/server.js:758:26)
    at Socket.emit (events.js:107:17)
    at Socket.onevent (/tmp/LaserWeb/node_modules/socket.io/lib/socket.js:330:8)
    at Socket.onpacket (/tmp/LaserWeb/node_modules/socket.io/lib/socket.js:290:12)
    at Client.ondecoded (/tmp/LaserWeb/node_modules/socket.io/lib/client.js:193:14)
    at Decoder.Emitter.emit (/tmp/LaserWeb/node_modules/socket.io/node_modules/socket.io-parser/node_modules/component-emitter/index.js:134:20)
    at Decoder.add (/tmp/LaserWeb/node_modules/socket.io/node_modules/socket.io-parser/index.js:247:12)
    at Client.ondata (/tmp/LaserWeb/node_modules/socket.io/lib/client.js:175:18)
    at Socket.emit (events.js:107:17)
    at Socket.onPacket (/tmp/LaserWeb/node_modules/socket.io/node_modules/engine.io/lib/socket.js:99:14)

<snip>
        socket.on('usePort', function (data) {
751: console.log('usePort data was ' +data);
                console.log(
</snip>

i also tried with Firefox 44.0 and Chromium: 45.0.2454.101, same results, tomorrow ill try with another machine, and another spare board

openhardwarecoza, 2016-01-31:
yeah, see we are getting null in the data, which explains why we also get an undefined on the next line trying to use said 'data'

So, issue isnt in server.js - lets check the next level up: 
Ctrl+shift+q in chrome, any errors? 

We can add some debug to see what are sending:
In /i/main.js:

https://github.com/openhardwarecoza/LaserWeb/blob/master/i/main.js#L559

Add a 
console.log('UsePort triggered - we selected: ' + $('#choosePort').val());

mrgastoso, 2016-01-31:
I added the code, and nothings show in the console log, seems that the function is not triggered when i select the port in the dropdown box

openhardwarecoza, 2016-01-31:
No other errors right after reload?

mrgastoso, 2016-01-31:
my fault, i was not watching the right console:

- at load:
SOCKET CONNECT 
main.js:562 UsePort triggered - we selected: null

- at port change dropbox
UsePort triggered - we selected: 32

openhardwarecoza, 2016-01-31:
Calling it a night.  Sorry,  work in a few hours (: 
Will check tomorrow,  in mean time,  try another machine.  When that selected 32 was sent,  did server.js 
Log 32 too?

mrgastoso, 2016-01-31:
Well, i think is ch340 driver related, i hooked an ftdi232 to the tx-rx pins of the mega board, and now it works with both ttyUSB0 and ttyUSB1 ports, extrangely it doesn work with ftdi alone

ERROR: Error connecting to webcam: connect ECONNREFUSED - Disabling Webcam Widget
WARN: Ignored:  Port /dev/ttyUSB1 said:  start
INFO:  Found device:  FTDI /dev/ttyUSB1 Firmware Detected: Marlin 1.0.0   Port ID No:  33
WARN: Ignored:  Port /dev/ttyUSB1 said:  echo:Marlin 1.0.0
WARN: Ignored:  Port /dev/ttyUSB0 said:  start
INFO:  Found device:  1a86 /dev/ttyUSB0 Firmware Detected: Marlin 1.0.0   Port ID No:  32
WARN: Ignored:  Port /dev/ttyUSB0 said:  echo:Marlin 1.0.0
WARN: Ignored:  Port /dev/ttyUSB1 said:  echo: Last Updated: Feb  1 2016 00:43:37 | Author: (none, default config)
WARN: Ignored:  Port /dev/ttyUSB1 said:  Compiled: Feb  1 2016
WARN: Ignored:  Port /dev/ttyUSB0 said:  echo: Last Updated: Feb  1 2016 00:43:37 | Author: (none, default config)
WARN: Ignored:  Port /dev/ttyUSB0 said:  Compiled: Feb  1 2016
WARN: Ignored:  Port /dev/ttyUSB1 said:  echo: Free Memory: 1910  PlannerBufferBytes: 2576
WARN: Ignored:  Port /dev/ttyUSB0 said:  echo: Free Memory: 1910  PlannerBufferBytes: 2576
WARN: Ignored:  Port /dev/ttyUSB1 said:  echo:Hardcoded Default Settings Loaded
WARN: Ignored:  Port /dev/ttyUSB0 said:  echo:Hardcoded Default Settings Loaded
WARN: Ignored:  Port /dev/ttyUSB1 said:  echo:Steps per unit:
WARN: Ignored:  Port /dev/ttyUSB0 said:  echo:Steps per unit:
WARN: Ignored:  Port /dev/ttyUSB1 said:  echo:  M92 X80.00 Y80.00 Z80.00 E0.00
WARN: Ignored:  Port /dev/ttyUSB0 said:  echo:  M92 X80.00 Y80.00 Z80.00 E0.00
WARN: Ignored:  Port /dev/ttyUSB1 said:  echo:Maximum feedrates (mm/s):
WARN: Ignored:  Port /dev/ttyUSB0 said:  echo:Maximum feedrates (mm/s):
WARN: Ignored:  Port /dev/ttyUSB1 said:  echo:  M203 X7600.00 Y7600.00 Z10.00 E25.00
WARN: Ignored:  Port /dev/ttyUSB0 said:  echo:  M203 X7600.00 Y7600.00 Z10.00 E25.00
WARN: Ignored:  Port /dev/ttyUSB1 said:  echo:Maximum Acceleration (mm/s2):
WARN: Ignored:  Port /dev/ttyUSB0 said:  echo:Maximum Acceleration (mm/s2):
WARN: Ignored:  Port /dev/ttyUSB1 said:  echo:  M201 X2600 Y2600 Z2 E2
WARN: Ignored:  Port /dev/ttyUSB0 said:  echo:  M201 X2600 Y2600 Z2 E2
WARN: Ignored:  Port /dev/ttyUSB1 said:  echo:Acceleration: S=acceleration, T=retract acceleration
WARN: Ignored:  Port /dev/ttyUSB0 said:  echo:Acceleration: S=acceleration, T=retract acceleration
WARN: Ignored:  Port /dev/ttyUSB1 said:  echo:  M204 S2000.00 T2000.00
WARN: Ignored:  Port /dev/ttyUSB0 said:  echo:  M204 S2000.00 T2000.00
WARN: Ignored:  Port /dev/ttyUSB1 said:  echo:Advanced variables: S=Min feedrate (mm/s), T=Min travel feedrate (mm/s), B=minimum segment time (ms), X=maximum XY jerk (mm/s),  Z=maximum Z jerk (mm/s),  E=maximum E jerk (mm/s)
WARN: Ignored:  Port /dev/ttyUSB0 said:  echo:Advanced variables: S=Min feedrate (mm/s), T=Min travel feedrate (mm/s), B=minimum segment time (ms), X=maximum XY jerk (mm/s),  Z=maximum Z jerk (mm/s),  E=maximum E jerk (mm/s)
WARN: Ignored:  Port /dev/ttyUSB1 said:  echo:  M205 S0.00 T0.00 B20000 X20.00 Z0.40 E5.00
WARN: Ignored:  Port /dev/ttyUSB0 said:  echo:  M205 S0.00 T0.00 B20000 X20.00 Z0.40 E5.00
WARN: Ignored:  Port /dev/ttyUSB1 said:  echo:Home offset (mm):
WARN: Ignored:  Port /dev/ttyUSB0 said:  echo:Home offset (mm):
WARN: Ignored:  Port /dev/ttyUSB1 said:  echo:  M206 X0.00 Y0.00 Z0.00
WARN: Ignored:  Port /dev/ttyUSB0 said:  echo:  M206 X0.00 Y0.00 Z0.00
WARN: Ignored:  Port /dev/ttyUSB1 said:  echo:Laser lifetime usage:
WARN: Ignored:  Port /dev/ttyUSB1 said:  echo: Hours: 0
WARN: Ignored:  Port /dev/ttyUSB0 said:  echo:Laser lifetime usage:
WARN: Ignored:  Port /dev/ttyUSB0 said:  echo: Hours: 0
WARN: Ignored:  Port /dev/ttyUSB1 said:   Minutes: 0
WARN: Ignored:  Port /dev/ttyUSB1 said:  echo:PID settings:
WARN: Ignored:  Port /dev/ttyUSB0 said:   Minutes: 0
WARN: Ignored:  Port /dev/ttyUSB0 said:  echo:PID settings:
WARN: Ignored:  Port /dev/ttyUSB1 said:  echo:   M301 P22.20 I1.08 D114.00
WARN: Ignored:  Port /dev/ttyUSB0 said:  echo:   M301 P22.20 I1.08 D114.00
WARN: Ignored:  Port /dev/ttyUSB1 said:  echo:SD init fail
WARN: Ignored:  Port /dev/ttyUSB0 said:  echo:SD init fail
INFO: Machine Profiles read from file

Checking for Updates
INFO: Git Refresh Fetching origin

INFO: Git Refresh En la rama master
Su rama está actualizada con «origin/master».
nothing to commit, working directory clean

INFO:  Now interacting with  /dev/ttyUSB0  running  Marlin 1.0.0
INFO:  Now interacting with  /dev/ttyUSB1  running  Marlin 1.0.0
INFO:  Now interacting with  /dev/ttyUSB0  running  Marlin 1.0.0

openhardwarecoza, 2016-02-01:
Excellent!  So we can close this one?   There is an open issue here to improve the serial bridge - when we get tonit,  i will make sure to test ch340

closed 2016-02-01 by openhardwarecoza

mrgastoso, 2016-02-01:
the problem is not solved, only i have tested in an unapropiate way, using the primary serial port the app cant connect

openhardwarecoza, 2016-02-01:
Works for me, so keeping it closed. Isolate whats wrong on your machine...