LaserWeb/deprecated-LaserWeb1 #98: Smoothie moving slow and multiple decimals in S command

cojarbi, 2016-02-02:
since grbl is not working im back at smoothie.

SVG works without issues ( should add Z movement option )

CALIBRATION: ( 2000 FR)

Movement is very slow no matter whats setting for feedrate
Power is Off all the time
17 decimals in S command

Attached gcode
[smoothie calibration GCODE-LaserWeb (2).gcode.TXT](https://github.com/openhardwarecoza/LaserWeb/files/113367/smoothie.calibration.GCODE-LaserWeb.2.gcode.TXT)
code for calibration

RASTER: seems to be ok

funinthefalls, 2016-02-02:
Doesn't seem to be a problem with svg2gcode.js. The grbl and marlin code appears to be the same.

From svg2code.js:
  // Firmware Specific Gcode Output
	if (firmware.indexOf('Grbl') == 0) {
		intensity = svglaserpwr.map(0, 100, 0, 255);
		//console.log('Mapping Intensity range for Grbl S0-S255');
		intensity = intensity.toFixed(0);
}
 else {
		intensity = svglaserpwr.map(0, 100, 0, 100);
		//console.log('Mapping Intensity range for S0-S100');
		intensity = intensity.toFixed(0);
	}

Where else do you post-process the Gcode?

cojarbi, 2016-02-02:
For smoothie no where else

funinthefalls, 2016-02-02:
OOps, missed that you were dealing with smoothie now, not grbl.

else if (firmware.indexOf('Smooth') == 0) {
		intensity = svglaserpwr.map(0, 100, 0, 1);
		//console.log('Mapping Intensity range for Smoothieware S0-S1');
		intensity = intensity.toFixed(2);
	}

setting the intensity seems to be different for smoothie, all the others use "intensity = intensity.**toFixed(0)**;" and smoothie uses "intensity = intensity.**toFixed(2)**

Could this be the problem?

funinthefalls, 2016-02-02:
Also in the raster section it sets all the other firmware to "**intensity.toFixed(0)**" but sets smoothie to "**intensity.toFixed(1)**"
Was setting "**intensity = intensity.toFixed(2)**" meant to be "**intensity = intensity.toFixed(1)**"

openhardwarecoza, 2016-02-02:
Smoothie, being a 32 bit board, isnt limeted to 255 levels of gray (;
S0 - S1 can even do 3 decimals.  We set it to 1 to only have 10 levels - makes the optimisation very fast (albeit at 10 gray levels instead of full grayscale)

cojarbi, 2016-02-02:
Ok so that's not an issue then.  But why I'm getting no power bellow 80%? There must be something specific to smoothie since it works in grbl

openhardwarecoza, 2016-02-03:
According to http://www.shapeoko.com/forum/viewtopic.php?t=7559&p=59718 the Jtech lasers need a specific PWM frequency.  

In your smoothie config, what is your:

`#laser_module_pwm_period                      20              # this sets the pwm frequency as the period in microseconds`

openhardwarecoza, 2016-02-03:
Looks like it needs 250khz...
20 microseconds = 500khz? 
So try 
`#laser_module_pwm_period 40`

cojarbi, 2016-02-03:
tested now is completely off. no power at all. So this means we need to go bellow 20?

openhardwarecoza, 2016-02-03:
Longer microseconds will mean lower frequency.  So try 2000?

cojarbi, 2016-02-03:
Disregard my last statement. Something else is wrong. Now not even 20 is working. Last night it did. So I must have another issue. Will troubleshoot and restart testing.

cojarbi, 2016-02-04:
So 200 seems to be it. Should we add something to the wiki

closed 2016-02-04 by cojarbi

openhardwarecoza, 2016-02-04:
Yes please.  Add a page for JTech

cojarbi, 2016-02-04:
[image]