LaserWeb/deprecated-LaserWeb1 #99: Smoothie Rastering Error

saskia-fabcreator, 2016-02-02:
Am trying to raster engrave using smoothieboard, it worked happily before the weekend but coming up with an error after update:

Application Error: ReferenceError: intensity is not defined (from http://145.107.25.149:8000/raster/laserraster.js line 209

```
207   var speed = this.config.feedRate;
208   if (lastGrey != this.grayLevel) {
209   intensity = this.figureIntensity();
210   speed = this.figureSpeed(lastGrey);
211   lastGrey = this.grayLevel;
```

cojarbi, 2016-02-02:
What kind of laser r u using?

saskia-fabcreator, 2016-02-02:
40W self built smoothieboard controlled, connected to laptop

openhardwarecoza, 2016-02-02:
Just tested, on a smoothieboard, and working fine for me

Do a git pull if you havent yet

saskia-fabcreator, 2016-02-04:
Yes, it seems to be happy again

closed 2016-02-04 by saskia-fabcreator