LaserWeb/deprecated-LaserWeb1 #100: Experimental adaptive feedrate on raster while unchaked

cojarbi, 2016-02-04:
Raster is generating speed variation even though the setting has not been chosen.
[image]
[GCODE-LaserWeb (1).gcode.txt](https://github.com/openhardwarecoza/LaserWeb/files/116843/GCODE-LaserWeb.1.gcode.txt)

openhardwarecoza, 2016-02-04:
@nathanielstenzel  are you handling the unticking too?  I think i saw this too,  but after generating a job with adaptive ticked,  then opening a second file and unticking.   @cojarbi or is it on first open for you?

cojarbi, 2016-02-04:
First open.  Freshly started server and loaded page.

nathanielstenzel, 2016-02-04:
I am positive this was working. The checked/unchecked status is not getting through. The last time I tried adjusting anything, I could not get a check box status to get through to laserraster.js. I did the same thing that the other checkbox was doing. It did not work. I searched online and tried something else and still no luck. I thought I was either going crazy or needed sleep. This was a week or two ago. I am surprised that nobody would have complained during that time period. As for me, I have not committed any code else than the one that was rolled back via someone's pull/merge and then the re-inserting of it directly into your code. That was the end-line optimization code. The button I tried to add in was a toggle for end-line optimization.

nathanielstenzel, 2016-02-04:
this.config.useVariableSpeed should be true or false in laserraster.js.

cojarbi, 2016-02-07:
@nathanielstenzel can you fix this please, please, please? It slowing me down. I cannot get a full power or constant speed on raster.

openhardwarecoza, 2016-02-09:
I can reproduce the issue...

openhardwarecoza, 2016-02-09:
Found the bug...   

http://heyjavascript.com/javascript-string-to-boolean/ told me how to fix it (;

closed 2016-02-09 by openhardwarecoza